c3d/db48x #1105: π and e constants (e.g. written in an expression) cannot be converted to a numerical end result.

prismv, 2024-08-11:
'cos π' and 'ln e' cannot converted to a numerical result with `→Num`. The former gives an error (result exp2 and cos error: Bad argument type).

All constants I tried (like i j inf g c ...) are converted to float (partially with unit). So e.g. '1/sqrt(epsilon_0 mu_0)' gives c as float and unit (should be m/s, but H and F expressions are not converted to 1/s).

Btw I was not able to reopen the same issue. User rights are just commenting.

              Closing as "won't fix" for now.
@prismv, please reopen if my answers are not satisfactory.

_Originally posted by @c3d in https://github.com/c3d/db48x/issues/1102#issuecomment-2282490589_

c3d, 2024-08-11:
It's strange, because the examples that fail for you (e.g. `ln e`) work fine for me, but the `1/sqrt(epsilon_0 mu_0)` case fails for me.

So that seems to suggest there is something broken in the constant evaluation. I'll look into it.

Wiljea, 2024-08-12:
@c3d : Same thing for me, there is no bug in the evaluation of math csts **e** and **π**. There is no issue.
The other problem with **1/SQRT(ε0ˑμ0)** is not with the numerical value of **c** (correctly evaluated here) but is due to the missing simplification of physical units:
**(1ˍFˑ1ˍH)^0,5=1ˍm/s**   OR   **1ˍFˑ1ˍH=(1ˍm/s)^2**.

prismv, 2024-08-12:
@c3d @Wiljea Thank you for your feedback. If I remember correct I had updated from dev branch with `make OPT=small` (due to issue #1066) and Setup->About showed the right version v0.7.13Z. Nevertheless I load QSPI and program from release packange and again from stable branch with `make OPT=small`. Both worked as expected, i.e. with no bug for conversion of π and e. I have no clue why the first firmware upload did not work.

Thanks to register #1110.

closed 2024-08-12 by prismv