c3d/db48x #1110: Evaluating `1/sqrt(epsilon0*mu0)` with variable `M` set

c3d, 2024-08-12:
This recalls the value of variable `M` and evaluates it, which is wrong.

Wiljea, 2024-08-12:
It seems to me that the problem with **1/SQRT(ε0ˑμ0)** is not with the numerical value of **c** (correctly evaluated here) but only with the missing simplification of physical units:
**(1ˍFˑ1ˍH)^0,5=1ˍm/s**   OR   **1ˍFˑ1ˍH=(1ˍm/s)^2**.
Possible solution may be to attach an equivalence to either definition of **1ˍF** or **1ˍH**:
**1ˍH=(1ˍm/s)^2/1ˍF**  OR  **1ˍF=(1ˍm/s)^2/1ˍH**
Was it what you meant here ?

prismv, 2024-08-12:
Yes, the value is right in the example given in #1105, but the units $\sqrt{\mathrm{F}\cdot\mathrm{H}}$ are not converted to most basic SI unit: s.

Though I don't understand the issue headline ...

prismv, 2024-08-15:
I tried the current dev version 0709be1.
Some factoring issue is solved now by showing {c}m/(F<sup>½</sup>H<sup>½</sup>). I see also the point that with strict compatibility to HP48/50 one has to perform UBase ("Base") manually. 

Nevertheless it might be a nice optional (?) feature to perform this last step automatically?!

commit 0709be13 (2024-08-14): units: Correctly factor out non-integral powers

When we evaluate an expression like `1/√(ε₀·µ₀)`, we compute the
square root of unit `F*H`. Unit `F*H` evaluates as `s^2`, so this
ends up with `s`.

This requires evaluating `F^(1/2)*H^(1/2)`, which needs unit mode
flags (`unit::mode` and `unit::factoring`) to be correctly set along
the way, notably while computing powers.

With this, evaluating `'1/√(ε₀·µ₀)' →Num` gives the correct, albeit
unsimplified result `299792458_m/(F^(1/2)*H^(1/2))`. This seems
consistent with the way the HP50G does it. Using `UBase` gives the
correct simplified result [... cut]
commit f2078057 (2024-08-16): units: Correctly factor out non-integral powers

When we evaluate an expression like `1/√(ε₀·µ₀)`, we compute the
square root of unit `F*H`. Unit `F*H` evaluates as `s^2`, so this
ends up with `s`.

This requires evaluating `F^(1/2)*H^(1/2)`, which needs unit mode
flags (`unit::mode` and `unit::factoring`) to be correctly set along
the way, notably while computing powers.

With this, evaluating `'1/√(ε₀·µ₀)' →Num` gives the correct, albeit
unsimplified result `299792458_m/(F^(1/2)*H^(1/2))`. This seems
consistent with the way the HP50G does it. Using `UBase` gives the
correct simplified result [... cut]
commit aca1c174 (2024-08-17): units: Correctly factor out non-integral powers

When we evaluate an expression like `1/√(ε₀·µ₀)`, we compute the
square root of unit `F*H`. Unit `F*H` evaluates as `s^2`, so this
ends up with `s`.

This requires evaluating `F^(1/2)*H^(1/2)`, which needs unit mode
flags (`unit::mode` and `unit::factoring`) to be correctly set along
the way, notably while computing powers.

With this, evaluating `'1/√(ε₀·µ₀)' →Num` gives the correct, albeit
unsimplified result `299792458_m/(F^(1/2)*H^(1/2))`. This seems
consistent with the way the HP50G does it. Using `UBase` gives the
correct simplified result [... cut]
closed 2024-08-18 by c3d