c3d/db48x #1118: [App] Increase key hit box sizes

mahi2003, 2024-08-19:
This issue is about the iOS app

I seem to miss key presses a lot more often than for example in [iHP48](https://apps.apple.com/us/app/ihp48/id1549608953) on my iPhone 12 mini. I think the problem is that the key hit boxes are too small. Sure, they follow the outline of the keys, but the keys are small on a 12 mini and the space between the keys serves no purpose. In my opinion it would be better to expand the hit box size in all directions so that there is no inactive space between the keys.

The hit box does not have to respect the labels around the keys. The actual key should be centered in the hit box, as that is what you are aiming for with your finger. So, it's okay when a key's hit box includes a (small) portion of the labels of the key below or left from it.

c3d, 2024-08-19:
Interesting suggestion. I'll try that.

mahi2003, 2024-08-21:
I just tried the new app version (0.7.14), and it's much better now. Thanks!