c3d/db48x #1120: [App] Lag on simple actions that is not present on the physical calculator

mahi2003, 2024-08-19:
This issue is about the iOS app

DB48X on an iPhone gives a massive performance boost over the DM32/DM42. However, some simple actions seem to have a delay that is not present on the DM32/DM42.

* Start from a clean state (stack empty)
* Press the ´1´ key → Near instant response
* Press `ENTER` to push the value to the stack → 1.5 seconds delay
* Press `←` to drop the value from the stack → 1.5 seconds delay

Also, when launching the app, the calculator display remains black for about 3 seconds. However, if I press a key during that time the display immediately appears.

c3d, 2024-08-19:
Indeed, this is related to the aggressive display optimizations on DM32/DM42 to reduce battery drainage. Some screen refresh events are related to events (mostly timer) that do not exist on the iOS app. The plan is to inject additional dummy events to cause the refresh.

c3d, 2024-08-22:
To confirm that the problem is a screen refresh, can you check if, when you hit the problem, the keyboard is still responsive (and causes a refresh)?

c3d, 2024-08-23:
Installed the version from the app store, and the problem is glaringly obvious. It does not appear when I locally install. I believe that the difference is debug vs. "archive" (release) builds. That narrows down where the problem might lie.

c3d, 2024-09-08:
This should be fixed in the version now distributed on the App store (starting with 0.7.16). Closing the issue, @mahi2003 please feel free to reopen if the problem is still there.

closed 2024-09-08 by c3d