c3d/db48x #1126: To slow

Uli-Moin, 2024-08-20:
Hi,
I am using db48x on my iPhone 15 Pro and iPad Pro.
In both cases calculating is very slow.
I.e. 6 enter 8 enter + needs nearly one second.

Uli-Moin, 2024-08-20:
Correction: it needs 2 sec !

mahi2003, 2024-08-21:
Same issue as #1120.

c3d, 2024-08-22:
Thanks for reporting this. Can you confirm which version you observe this on? (use the `VERSION` command).

Also, as far as I can tell, it is not really a speed problem but a screen refresh problem. There is a condition which I did not fully identify yet which causes the screen to not be refreshed after a stack update.

c3d, 2024-08-22:
Also, at the moment, this does not reproduce on any of my devices, which makes it a bit more difficult to fix ;-)

Uli-Moin, 2024-08-22:
HiIt is Version 0.7.14. Mit freundlichen GrüßenUli*

Uli-Moin, 2024-08-22:
> Too confirm if the problem is a screen refresh issue as I think, can you please check that when you hit the problem, the keyboard remains responsive (and should also cause a screen refresh, e.g. if you type '1' you should see the editor pop up).

Yes it is!!!If I press 3 Enter5 Enter+And when I press suddenly after „+“ the „1“ the result „8“ comes in this moment on the screen.  Mit freundlichen GrüßenUli*

Uli-Moin, 2024-08-22:
Yes!When I start the app the screen needs seconds to become white. But when I start the app it becomes white suddenly as soon as I press „1“. Hope it helps. Mit freundlichen GrüßenUli*Mit freundlichen GrüßenUli Rackisch*

c3d, 2024-08-23:
Installed the version from the app store, and the problem happens every single time. With my local builds of the app, they do not. That seems to narrow it down to some difference between debug and release build.

c3d, 2024-08-23:
Marking it as a duplicate of #1120

c3d, 2024-08-23:
OK, now my app store version does not show it anymore.

Also added schemes to my iOS builds to be able to build release and debug, and that's not it.

c3d, 2024-09-08:
This should be fixed now, closing that issue.

closed 2024-09-08 by c3d