c3d/db48x #1127: History Navigation

treepble, 2024-08-21:
posted as requested.

Navigating in History using the **'<-word'** and **'word->'** soft keys(sk) act differently.
The **'<-word'** sk moves the cursor to a new object back through the history list with every single push, as expected while
the **'word->'** sk moves the cursor forward through history items with two pushes, the first push moves the cursor from object left side to object right side the the second push moves on to the next list object.

I watched your video and I know your aware of this and I'm probably missing the reason but it just feels wrong.

Thanks.

c3d, 2024-08-22:
Reopening, since I want to address it (@treepble, not sure why you closed it right away?)

c3d, 2024-11-05:
I think this was fixed now. @treepble can you please check?