c3d/db48x #1131: Rounding flags

c3d, 2024-08-22:
Add rounding flags for decimal computations, to be able to round up, down or towards/away from zero.

Wiljea, 2025-04-02:
Are these rounding flags what would allow to implement the operations +inf +sup *inf *sup which are suggested in the French article on interval arithmetic: https://fr.wikipedia.org/wiki/Arithm%C3%A9tique_d%27étapes? Furthermore, are these functions currently defined? And, if so, how are they used or defined?

c3d, 2025-04-02:
No, the function are not implemented yet, as indicated by the fact that the issue is still open. Furthermore, a quick study of what it would take to implement them highlighted that it was not that simple.