c3d/db48x #1138: For a new formula in Optics the function sinc(x)=sin(x)/x is required

Wiljea, 2024-08-24:
This function doesn't exist either in the HP50g Its limit is 1 at x=0 (does the limit function is already defined in db48x ? It doesn't seem. No need for the moment, anyway.) 
SOLUTION
To avoid the singular calculation at X=0 the following RPL snippet replace the fonction by its McLaurin series near that point. Suppose X_r :
« →X 
    « if X abs 0.1 ≤ 
       then
           1  X   x↑2  6  /  -
       else
           X   SIN   X   /
       end
    »
»

c3d, 2024-08-31:
What does the whole formula look like? Is it problematic to actually write `sin(x)/x` in the equation?

I will try to replicate your observations about `plot`.

Wiljea, 2024-09-01:
The whole formula is:
(I_(W/m^2))/(Imax_(W/m^2))=(SIN(α_r)/α_r)^2'
or
(I_(W/m^2))/(Imax_(W/m^2))=(SINC(α_r))^2'

Wiljea, 2024-09-01:
The whole formula is:
(I_(W/m^2))/(Imax_(W/m^2))=(SIN(α_r)/α_r)^2'
or
(I_(W/m^2))/(Imax_(W/m^2))=(SINC(α_r))^2'
It is problematic if the user choose (α_r) = 0