c3d/db48x #1139: The db48x needs the Black-Body Integral function F0λ(T_K, λ_nm) (Low priority)

Wiljea, 2024-08-24:
The db48x needs the emissive power Black-Body Integral function F0λ(T_K, λ_nm) defined in the HP50G (HP50G_AUR 3-82 and 5-31 for a graphic illustration of the under the curve surface). This is a low priority issue since there is only one section "Black Body Radiation" concerned in the chapter "Heat Transfer".

Wiljea, 2024-09-11:
Either the developer defines the function F0λ(T_K, λ_nm) internally or I calculate them directly in the Equation Library. 
We need the precise algebraic syntax of the n-ary function INTEGRATE. Is it: INTEGRATE(f(x); x; xmin; xmax) ? Let's assume this syntax is the correct one.

Since the integral from 0 to ∞ of g(x) = x^3/(exp(x)-1) = π^2/15, one can verify on the db48x that the integral of f(x) = (15/π^2)·x^3/(exp(x)-1) gives the normalized value of one. Concretely: 

with x = Ⓒh·Ⓒc/((λ_nm)·Ⓒk·(T_K)) one can verify numerically:

INTEGRATE(f(x); x; 0.001; 1000) = 0.99999 99999 31043 which is close enough to 1

Therefore a call to the emissive power function will translate as:

F0λ(T_K, λ_nm) = 
INTEGRATE((15/π^2)·x^3/(exp(x)-1); x; Ⓒh·Ⓒc/((λ_nm)·Ⓒk·(T_K)); 1000)

LdBeth, 2025-03-11:
I think the equation used is 

[image]

Which verifies to f = F0λ(λ2;T) – F0λ(λ1;T) = (1-(15/π^4)*Integrate[x^3/e^x-1,{x,0,11.3022535546}])-(1-(15/π^4)*Integrate[x^3/e^x-1,{x,0,18.8370892577}])

With Wolfram alpha, it is 0.00361346 - 7.98896e-6 approximately 0.0036

Wiljea, 2025-03-11:
Is there still a problem remaining (I don't think so with the recent revision of the Romberg procedure by c3d)? If not, I will close this issue.

LdBeth, 2025-03-11:
Neither do I think there is still a problem. I'll raise a different issue on giving up numberic integration.

Wiljea, 2025-03-12:
Done !

closed 2025-03-12 by Wiljea