c3d/db48x #1143: Alpha-Shift keyboard entry issue

treepble, 2024-08-30:
There is an issue with the alpha-shift(orange key) keyboard entry.

**Tested on DM32 with V0.7.14 and V0.7.15**

If you turn on the calculator and clear the screen (ON key)
then alpha-shift the numeric key 4, you will mostly get 't'
but if you keep the shift key held and keep hitting the 4 key you will
occasionally get 4 and sometimes no entry, like the key press has been missed.
If you don't see this after a line of characters keep holding the alpha-shift key down, 
erase all the characters with(<-) key and start again alternating between the shift 4 and 5 keys.
This issue appears to be confined to the 4 key however I've not tried every key.

Also a related issue is: from a power up and screen clear, if you press the alpha-shift key
without a follow up key the calculator beeps and if you hold the alpha-shift key
the calculator just keeps beeping.

c3d, 2024-09-01:
Hello @treepble, I have trouble reproducing your issue.

I believe that what you call "alpha shift" is what is the yellow key on the original DM32, and the right arrow key [on the DB48X keyboard](https://github.com/c3d/db48x/blob/dev/Keyboard-Layout.png) with "edit" next to it. Is that correct? If that is the case, please understand that when running DM32, this is no longer a "shift" or "alpha shift", but the right arrow key. I strongly recommend you [get a keyboard overlay](https://www.hpmuseum.org/forum/thread-20113.html).

Now, if my interpretation is correct, then I believe that all the behaviours you describe are intentional and correct, although they may need an explanation. The detailed explanation is [in the documentation](https://github.com/c3d/db48x/blob/dev/help/db50x.md#alpha-mode).

> If you turn on the calculator and clear the screen (ON key) then alpha-shift
> the numeric key 4, you will mostly get 't' but if you keep the shift key held
> and keep hitting the 4 key you will occasionally get 4 and sometimes no entry,
> like the key press has been missed.

The left and right keys are used for what is called _transient alpha_, intended to facilitate the quick entry of alphabetic characters because they are so important in RPL.

The problem with transient alpha is that it's very timing sensitive and therefore requires a bit of training. That's because it overloads the arrow keys to do two completely different things: acting as an arrow key or selecting uppercase (left arrow) or lowercase (right arrow) alphabetic characters.

The algorithm used is that you need to press the arrow and then immediately afterwards, the letter key. If you do not press the second key quickly enough, then the first key is interpreted as an arrow.

It requires a bit of practice, but once you get it, it's really fast and convenient. Once transient alpha is enabled, it should remain until you release the arrow key, i.e. the other keys should show as alpha characters only.

> If you don't see this after a line of characters keep holding the alpha-shift
> key down, erase all the characters with(<-) key and start again alternating
> between the shift 4 and 5 keys.  This issue appears to be confined to the 4
> key however I've not tried every key.

Now, another possible problem is that your 4 key may be a bit sticky or that the underlying keyboard might be shifting a bit. So maybe when you hit 4, which is close to the right arrow, you end up pressing hard enough that the arrow key loses contact? That would explain how you see 4 appear in the middle. As written above, that should not happen as soon has you have enabled transient alpha, and that seems to indicate that the right arrow key was seen as released, even if you did not intend to.

In my experience, keys on DM32 and DM42 calculators can sometimes be a bit capricious. On my DM32, the C key was so unreliable that I ended up replacing the contact sheet in the calculator (SwissMicros sent that to me). Can you run a self-test on your keyboard to see if there is an issue with either the arrow or the 4 key? To get the self-test, go to SETUP, then 8. System, then 2. Enter System Menu, then 8. Self Test, then 1. KBD Test.

You should have something that says:

```
Production Diagnostics
Press keys by rows

[38] 00 - 00

Press [+] and [-] to exit
```

The `[38]` is the key that the self-test expects you to press (in that case the first function key in the top-left of the keyboard). That number may change if you run the self-test "the right way" by pressing the keys it wants. For now, you can ignore it for the test I want you to do here.

On the right of `[38]`, the first `00` is the key code of the first key you press. The second `00` is the second key code when you press two keys at the same time.

Then do the following test, where RIGHT is the DB48X right arrow key which I think you called alpha shift, and LEFT is the DB48X left arrow key corresponding to XEQ on the DM32:

1) Press LEFT. The display should change to `[38] 18 - 00`. The `18` is the key code for the LEFT key. That should return to `00` when you release the LEFT key.

2) Press RIGHT. The display should change to `[38] 23 - 00`. If you see occasional cases where `23` does not show up when you press the key, then maybe you have a problem here.

3) Press and hold LEFT, then 7. You should see `[38] 19 - 18`. That indicates that you pressed 7 last, and that you are holding LEFT. If you release LEFT and keep 7 pressed, you should see `[38] 19 - 00`. If you release 7 and keep LEFT pressed, you should see `[38] 18 - 00`.

4) If you do the same with RIGHT and 4. You should get `[38] 24 - 23` and be able to reliably get either `[38] 24 - 00` if you release RIGHT first, and `[38] 23 - 00` if you release 4 first.

If any of these tests fails or seem unreliable, then that may explain part of the issues you have. I would recommend to contact SwissMicros to ask if they can help with the keyboard contacts. I must admit that the test fails for me sometimes if the calculator is not directly on the table but placed on the puch holder and I do the test with only one hand.

> Also a related issue is: from a power up and screen clear, if you press the
> alpha-shift key without a follow up key the calculator beeps and if you hold
> the alpha-shift key the calculator just keeps beeping.

That's because if there is no other key, then after a very short while, it's interpreted as RIGHT. When you are not in the editor, that key brings the first level of the stack into the editor (that's the little "EDIT" on the right of the key). If there is nothing in the stack, then it beeps.

To confirm that this is indeed what happens, press 123 ENTER, then hit that key. You should see 123 appear in the editor with a cursor on the left.

treepble, 2024-09-02:
Hi Christophe,
thanks for the detailed explanation.
The calculator passed all the tests and I never noticed any issues with the keys on the original DM32 firmware, however I believe you must be right about the hardware as I've seen other strange actions over the last few days of testing. I'll restore the original DM32 firmware and give it a good workout. I might leave it as original and sometime in the future throw a few more dollars at SM and get another one.

Anyway good luck with the project and I look forward to seeing it when finished.

-=_Nick_

c3d, 2025-04-28:
Did not hear back from @treepble, closing for now.

closed 2025-04-28 by c3d