c3d/db48x #1163: Add usual units to the `Computing` units menus

c3d, 2024-09-03:
I can't remember the location, but someone commented that they expected units such as kilobytes, not just bytes.

ormandj, 2024-09-11:
Probably myself, here: https://www.hpmuseum.org/forum/thread-20157-post-191350.html#pid191350

Thank you!

c3d, 2024-09-11:
Thanks. Indeed, it was you, @ormandj. I'm glad my evil trick to get you to reveal your identity on GitHub worked, muhaahahaha!

More seriously, here are the changes coming in 0.7.17, which I would like to try to release during the night:

1. When entering a number and then selecting a unit, the unit value is not immediately pushed on the stack. Instead, it remains in the editor. This means that you can enter multiple units.

2. There is a new editor mode  when after a `_`, identified by a `U`-shaped cursor. In that mode, the `×` and `÷` keys do not immediately perform the operation, which makes it easier to enter units such as `m*s`.

3. In this mode, it is possible to chain units and inverse units. If you type the functions keys for `m` and then for second, you get `1_m·s`. If you then add the inverse for `km/h`, you end up with `1_m·s/(km/h)`.

4. When in unit mode, the `EEX` (or `×10ⁿ`) key cycles through prefix units. So if you entered `1_m` and then hit `EEX`, you first get `1_km`, then `1_Mm`, and so on. Note that this does not change the numerical value to match, it's only a data-entry facility.

All this should make it much easier to quickly enter a unit like `1_Gbit/s`. I'll let you test.

ormandj, 2024-09-11:
I need to purchase a calculator now, so I can use your firmware. I'm using your IOS application for the time being. I just signed up for an account on the SwissMicros forum to try and contact them and inquire about having custom keypads printed, since it feels problematic to stick on little vinyl stickers to keypads, and the DM32 seems like the current calculator to get (and requires significant keypad changes). I was trying to find a way to contact you to discuss, but it appears you have now fallen into my trap by my appearing to fall into your trap!

Let me know if there's a way to communicate directly, and we can go from there. I'm happy to use discord/slack/matrix/whatever works best for you. :) My email: ormandj@corenode.com. I'll be a happy tester for you, and can contribute, as well, once I have a HW solution.

commit 9e7c0c68 (2024-09-03): units: Add usual "big" units to `Computing` units menu

Add units such as kilobytes, etc to the units menu

Fixes: #1163
commit 58e55dab (2024-09-04): units: Add usual "big" units to `Computing` units menu

Add units such as kilobytes, etc to the units menu

Fixes: #1163
commit 6e9f73ac (2024-09-08): units: Add usual "big" units to `Computing` units menu

Add units such as kilobytes, etc to the units menu

Fixes: #1163
closed 2024-09-12 by c3d