c3d/db48x #1178: Saving state in iOS throws a failed message

nickapos, 2024-09-13:
While trying to save or load message in iOS I get save state failed message or load state failed. 
however a state file is created in the iPhone. 
not sure if the state file is corrupted because load also throws the failure message.

c3d, 2024-09-13:
Thanks for the report, @nickapos.

We are in the usual WorksForMe™ situation, so I guess I need to ask for more details.

Can you tell me:
- What application are you using as reported by going to the `SETUP` screen (SHIFT-0) and then selecting `2. About`
- What is the exact sequence of operations to get the message, in both the "save" and "load" cases?
- What is the exact error message? Ideally a screen capture. Asking because there is no exact match, and fuzzing matches different possibilities.
- What iPhone model you use, and what version of iOS?
- Where are you trying to save the file? Is it on iCloud or local storage?
- What happens if you enter `0` and  `"FILE"` on the stack, and then hit the `STO` key?

Thank you in advance for your answers.

nickapos, 2024-09-13:
Hi Christophe, here are the screenshots

[image]
[image]
[image]
I am using an iphone 15 plus with iOS version 17.6.1

nickapos, 2024-09-13:
When i type 0 and 'FILE' and hit STO then i get no errors. It looks like FILE variable gets value 0

c3d, 2024-09-13:
> When i type 0 and 'FILE' and hit STO then i get no errors. It looks like FILE
> variable gets value 0

Yes, but this is not the correct test case.

I asked about using `"FILE"` and not `'FILE'` (double quote instead of single quote). This stores in a file instead of a variable.

c3d, 2024-09-13:
In the screenshots, I see no iOS file dialog. Did you see that prior to seeing the screen you show?

nickapos, 2024-09-14:
Yes the dialog opens to allow you to type in the name of the state file for save or to,choose which file to load.

nickapos, 2024-09-14:
Apologies about the FILE test. I did it again and got the following
[image]

c3d, 2024-09-15:
> Apologies about the FILE test. I did it again and got the following 

Thanks. That's helpful, as it gives me the source of the problem.

The calculators these are modelled after (SwissMicros DM32 and DM42) can only open one file at a time. It's a limitation of their built-in firmware. So there is a lot of internal logic to close one file and open the next one in case I need to deal with two files at the same time, e.g. when processing constants in equations, the constants may have units, so you end up juggling with three files.

What the "Too many open files" message indicates is that this logic is somehow violated in your configuration. I need to explore the code to see how this could happen. If I can't find it, I will add some code in the next release to capture what could cause the problem.

I suspect that the problem might be with iCloud. When the calculator starts, it tries copying units.csv, config.csv and equations.csv, as well as the help files, from the application resources to a user-visible application directory. This is to allow you to edit these files.

When you use the Load state dialog, you told me you see the iOS dialog. My expectation is that it should be in a place called "state". Please try to verify all the expectations below, and tell me if any looks odd.

On the left, you should have a blue text (a button) that looks like `< DB48X`. If you go there, you should see four folders: config, data, help and state. You should be able to go one level up, and see folders for various applications. One level up, and it should show you a dialog with various places, including "iCloud Drive" and "iPhone" (with a text that depends on localization). Normally, the folders you saw before should be on your local storage.

If you go back to the DB48X folder, then "config", you should see 5 files (characters, constants, equations, library and units). If you modified any of these files yourself, please send me your copy, as this might play a role.

Inside the DB48X folder, there is also a "data" folder, which should contain your "FILE"  file if things worked properly. Because of the error above, chances are that the folder is empty. For me, it contains just FILE.

Inside the DB48X folder, there is also a "help" folder, with three files and an img folder.

Finally, inside the DB48X folder, there is a "state" folder. That's the one where interesting stuff can be found. For all the files that are in there (and primarily the one called `Demo1` per your screenshot), can you select the file, use the "Share" button, then "Copy", then paste as a comment in this issue?

My suspicion is that there is something in this file that sends me down a path that is leaving a file open that should be closed (probably an error path that I forgot to cover, even if I did my best to use C++ constructors/destructors to deal with this).

nickapos, 2024-09-16:
Hi Christophe, you are absolutely right in all of your assumptions, except the one that i might have modified any of these files. I did not know that i could modify anything so i have not done it.
Here are the screenshots from my phone and the Demo1 state you requested
[image]
[image]
[image]
[Demo1.zip](https://github.com/user-attachments/files/17010173/Demo1.zip)

I have zipped the state file because otherwise github does not allow me to submit it.

c3d, 2024-09-16:
Did you by any chance install two distinct versions of DB48X, e.g. DB40X and DB50X?

c3d, 2024-09-16:
The reason I asked is because I could reproduce somewhat similar symptoms by picking up the directory for another application, which has the same structure (e.g. trying to load the files from DB40X from DB50X).

Though the internal message is a permission denied error, I suspect that this could lead down a path where a file is not properly closed.

nickapos, 2024-09-17:
No I only got DB48X

nickapos, 2024-09-17:
This is interesting. I did a test. I was getting the same behaviour in my iPad. In both cases the app was installed at the same day and has been updated for a while automatically.
So I removed the app and reinstalled it. Now the problem is gone. I can save and load without this error.

c3d, 2024-09-18:
That seems to confirm my hunch that the problem is related to some permissions accessing the files. iOS does really bizarre things with the sandboxed paths. What shows as a nice DB48X folder on your local storage has an entirely different name internally, and is sandboxed so that one app cannot modify the data for another.

I added some code to show the actual OS-level error message in the message and the name of the file being accessed. That will be available in the next update.

closed 2024-09-29 by c3d

c3d, 2024-09-30:
I believe that I have found and fixed a number of issues that could result in the symptoms you observed. At the very least, you should now have better error messages such as "Permission denied" when a problem occurs.

This should become available in 0.8.0.

commit a11c2965 (2024-09-20): files: Improve error message in case of error

When there is some error saving or loading the state file, put
additional user-visible information, e.g. from `errno`.

Also avoid blocking file load/save on "Too many open files" error
while running a release simulator version. The recorder message is
enough to know during testing, and it should not prevent actual users
from loading or saving a file.

Fixes: #1178