c3d/db48x #1186: Fundamental constants need to be updated to CODATA2022 (to be up to date)

Wiljea, 2024-09-15:
For the most recent revision **CODATA2022**, the reference for the extensive listing is 
https://physics.nist.gov/cuu/pdf/all.pdf

This concerns for instance basics like **h, hbar, k, NA, etc.** which are exactly defined (or calculated) since 2019 and are all actually in error in db48x (or HP50g). And possibly other constants as well. One should check each one carefully. For a start, I actually need the correct values for these 4 constants. I may try to check later for other value to be corrected. 
Thanks.

c3d, 2024-09-15:
Yes, the value that defines a constant can be any expression. Right now, this feature is only used to apply units. I think that an expression would parse correctly today, but I'm not positive it would be evaluated. If not, it's not hard to add.

Wiljea, 2024-09-15:
Wow, that's nice because for me its an important point to stress because, first this exploits the very nature of db48x which has variable precision, and second, an updated of secondary constants (the ones related to measured quantities) by adapting their equations will only need a modification of the CSV file in case of calculations changes or related constants modifications. Doing so, will make the db48x a very unique and unprecedented scientific calculator (even useful for CODATA officials).

Wiljea, 2024-12-23:
Done.

closed 2024-12-23 by Wiljea