c3d/db48x #1227: When trying to conver lb to kg or gr I am getting inconsistent units error

nickapos, 2024-10-01:
Today I realised that when I try to convert lb to kg I get inconsistent units error
[image]
Version is 0.8

c3d, 2024-10-02:
It's weird. It works for me, and works for the [on-line version](https://48calc.org). Could you please share:
1. The state file
2. The `units.csv` file

Is it possible that you have some customization in `units.csv`?

nickapos, 2024-10-02:
[units.csv](https://github.com/user-attachments/files/17220254/units.csv)
[Demo.zip](https://github.com/user-attachments/files/17220264/Demo.zip)

i think I added some entries in the cycle section, nothing else. Please let me know if you need anything else.

c3d, 2024-10-02:
With your file, I have the same problem. So now I can investigate.

There is another problem too: in the `Quick` menu, if I hit `Next` I get tons of garbage. Well, not exactly garbage, more precisely units from the builtin `Length` units.

c3d, 2024-10-30:
Fixed one of the issues (garbage in `Quick` menu), but other problems remain.

c3d, 2024-10-30:
The problem was that the `=Cycle` section of `units.csv` was incorrectly used for the definition of `lb`, leading to an invalid definition that caused the problem. This will be fixed by skipping the `=Cycle` section while looking for unit definitions.

commit c11dc23a (2024-10-31): units: Skip the `=Cycle` section for unit definitions

While searching for unit definitions, we incorrectly used the
"definitions" found in the `=Cycle` section of the file. This led to
incorrect `Inconsistent units` errors.

Skip the entire `=Cycle` section while looking for definitions.

Fixes: #1227
closed 2024-11-11 by c3d