c3d/db48x #1253: The functions written as ARCTAN, ARCCOS & ARCSIN need to be recognized

Wiljea, 2024-10-13:
Currently I am in the HELP file db48x.md and I want to write small snippet of RPL code to be executed before initiating the SOLVER and I wonder how to write ARCTAN, like in the following:
* Example 2: To show the infinite line approximation ('r << L') of the previous section we calculate [E_N/C;λ_C/m] with the angles θ1=**ARCTAN**(r/(L/2)) and θ2=360°-θ1 (see fig):
```
rpl
3 2 / 0.05 / ARCTAN 'θ1' STO 360° 'θ1' - 'θ2' STO  @ Expecting: [θ1=88.0876_°;θ2=271.9124_°]
@ Expecting [E=5.988364E5_N/C;λ=1.666667e-6_C/m]: OK. 
'ROOT([ⒺE FieldFiniteLine;Q=5E-6_C;r=0.05_m;∈r=1;L=3_m;θ1=88.0876_°;θ2=271.9124_°];[E;λ];[1_N/C;1_C/m])'
'E' 5.991701E5_N/C %Chg  @ Expecting [0,056%] of relative difference.
```

c3d, 2024-10-13:
The accepted spellings at the moment are `ATAN` (HP-compatible) and `tan⁻¹` (DB48X only).

It is not difficult to add the `arctan` spelling.

Wiljea, 2024-10-14:
ATAN will indeed do the job. I should have thought of that... I knew the notation tan⁻¹, but my problem was one of text editing in Notepad+ on my computer (Win10), what font should I use to have the exponent character ⁻¹ ? Anyway, I will close this issue eventually.

closed 2024-10-14 by Wiljea

c3d, 2024-10-14:
I just added the `arcsin`, `arccos` and `arctan` aliases.

commit f374cc0c (2024-10-14): parsing: Add `arcsin`, `arccos` and `arctan` spellings

Suggested by @Wiljea

Fixes: #1253
commit fdcdd729 (2024-10-15): parsing: Add `arcsin`, `arccos` and `arctan` spellings

Suggested by @Wiljea

Fixes: #1253