c3d/db48x #1262: Add assignment operations

c3d, 2024-10-17:
In order to be able to document the built-in equations in a way that is usable in practice, we need to have a quick way to describe a number of variables.

A possible approach is to have an `assignment` object that is similar internally to unit objects, but with a `=` separator and a name on the left. This would work like the `▶` operator on the HP50G.

Proposed syntax:

```
A=50_m
B='30_cm+25_mm'
```

Note that the syntax cannot be accepted inside algebraic expressions, but it could be rendered as:

```
50_m▶A
'(30_cm+25_mm)▶B'

```

This behaves like `STO` but returns the input value on the stack.

Wiljea, 2024-10-17:
So contrary to STO it doesn't consume the value on level 1 of the stack, isn't it ?

c3d, 2024-10-17:
Separate assignment operation and assignment object.

The assignment object also tags the result.

commit 230693f9 (2024-10-17): variables: Add `Copy` operation (`▶`)

The `Copy` operation corresponds to the `▶` operator on the HP50G,
which is documented as "▶ (Store)", but behaves differently from
regular `STO` in that it returns the stored value on the stack.

Fixes: #1262
commit 947ceeee (2024-10-17): variables: Add `Copy` operation (`▶`)

The `Copy` operation corresponds to the `▶` operator on the HP50G,
which is documented as "▶ (Store)", but behaves differently from
regular `STO` in that it returns the stored value on the stack.

Fixes: #1262
commit 76306702 (2024-10-18): variables: Add `Copy` operation (`▶`)

The `Copy` operation corresponds to the `▶` operator on the HP50G,
which is documented as "▶ (Store)", but behaves differently from
regular `STO` in that it returns the stored value on the stack.

Fixes: #1262
commit 3d9aa502 (2024-10-19): variables: Add `Copy` operation (`▶`)

The `Copy` operation corresponds to the `▶` operator on the HP50G,
which is documented as "▶ (Store)", but behaves differently from
regular `STO` in that it returns the stored value on the stack.

Fixes: #1262
closed 2024-10-21 by c3d