c3d/db48x #1264: From DEV: "∫ Error [Numerical precision lost]"

Wiljea, 2024-10-18:
This error condition appears in the simulator construction from DEV (V 0.8.1-22-db3b) after NEVAL#=4097 for the following integration program which is deliberately slowed down:
«
	0
	→ NEVAL
	«
		-0.2 12 PlotXRange
		-0.015 0.28 PlotYRange
		DrawAxes 1.⁳-10
		100 "Planck Distribution Integration"
		0.8+ⅈ0.28 DrawText
		"f" 11.5+ⅈ0.04 DrawText
		«
			→ x
			«
				'15÷Ⓒπ↑4·x↑3÷expm1
				x' Evaluate
				→ y
				«
					Ticks Foreground
					x 0 ℝ→ℂ x y
					ℝ→ℂ DrawLine
					#0 Foreground y 1
					'NEVAL' Store+
					if
						NEVAL 1 mod 0 =
					then
						"NEVAL# " NEVAL
						+ 5.7+ⅈ0.22
						DrawText 0.005
						Wait
					end
				»
			»
		»
		'x' ∫ Duplicate "∫ = "
		Swap + 6.1+ⅈ0.17
		DrawText 1 Swap - "Err: "
		Swap + 6.6+ⅈ0.12
		DrawText
	»
»

c3d, 2024-10-18:
This is an expected error message if the algorithm does not converge in the given number of iterations. The fact that you see it now may result from the fact that I bumped the desired precision from 12 to 18 digits by default.

I'll look into it.

LdBeth, 2025-03-11:
The numerical on the DM42n is giving up very early, say `-10 0 'EXP(X)' 'X' Integrate` can hang the calculator for a while then report  "Numerical precision lost", while the iOS 0.9.0 and Linux Sim 0.9.2 both gives 2.7182...

Wiljea, 2025-03-11:
Done