c3d/db48x #1265: Push stored value on the stack

c3d, 2024-10-18:
When using `3▶A`, we should put the stored value of `3` on the stack instead of the value used for initialization. That way, the stack reference does not required additional space, and the next GC can reclaim the temporary used for initialization.

Wiljea, 2024-10-18:
Just to clarify is it another notation for recall to be used in algebraic expression or RPL program? Besides using the CHAR key to access the "▶", how do you type it ? Is it really a simple right shift or some use of the transient F1...F6 ?

c3d, 2024-10-18:
The `▶` operator another notation for `Store`, not `Recall`. In the HP50G, it can only be used in algebraic mode as far as I know, i.e. between the horrible backticks of the algebraic mode.

### Comparison with HP50G

DB48X gets rids of backticks (did I mention this was a rather horrible design?) and makes the algebraic mode more full-featured. This is how `'ROOT(X=X^2-1;X;1)'` works (on the HP50G, it only works in algebraic mode, but not in algebraic expressions).

In line with that approach, `'3▶X'` is accepted in DB48X algebraic expressions, and behaves like `3▶X` in HP50G algebraic mode, i.e. it returns the value being stored. So `'42▶A+2'` stores 42 in `A` and returns `44`. Evaluation happens from left to right, so `'42▶A+2+A▶A+3'` parses as `'((42▶A)+2+A▶A)+3'`, stores `86` in `A` (`42+2+42`) and returns `89` (`86+3`).

If you are curious, the `▶` operator is (poorly) documented on page 3-303 of the [HP50G Advanced Reference Manual](https://literature.hpcalc.org/items/253).

### What is this issue about?

This particular issue is just describing a memory-saving optimization with `▶`, which ensures that the value on the stack does not use memory on its own. To explain how it works, I need to explain that in DB48X like on the original RPL, the stack is a set of pointers to objects, which can point to either temporary objects or global objects.

If you compute `1 2 +`: `1` and `2` are first created as temporary objects, then a `3` temporary object is created. At that point, if a garbage collection happens and if the `1` and `2` objects are no longer needed, they can be removed from memory.

If you store `3` in `A`, it creates a copy of the `3` object in the current directory, along with name `A`. If you recall variable `A`, the stack simply points to that global value, so it does not use any additional temporary space.

The first implementation of `▶` would create that global object, but then leave the original copy on the stack. This means that twice as much memory would be used. Not a big thing for a small object, but if you have a 20x20 matrix like you get with `20 IDN SIN`, that's 1063 bytes per copy.

commit b2e9ce0c (2024-10-19): variables: When using `3▶A`, put the stored `3` on the stack

When we use `3▶A`, we created a persistent global copy of a possibly
temporary stack object. This means that we have two copies of `3`, one
on the stack and one in globals. Since it's on the stack, the copy on
the stack would remain in case of garbage collection.

Use the stored value for the stack instead, so that the value on the
stack can be garbage collected. This matters if the object in question
is big, for example if we are manipulating an array.

Fixes: #1265
closed 2024-10-21 by c3d