c3d/db48x #1266: Replace tagged values for results with assignments

c3d, 2024-10-18:
Currently, the solver returns something like `A:0.5`, but @Wiljea suggested that we could return `A=0.5` instead. @Wiljea was thinking of returning equations, but we could use the newly created assignment object for that purpose.

c3d, 2024-10-18:
Probably necessary to also implicitly convert assignments to equations in `expression::get`.

c3d, 2024-10-18:
The primary practical benefit of returning an assignment is that it makes it possible to restore the value quickly using the interactive stack.

c3d, 2024-10-18:
In that context, it makes sense that the `SolvingMenu` keys also return assignments.

Wiljea, 2024-10-18:
For the purposes of some demo or of possible interactive solutions in the Library Equation, will it be a good idea to allow these affectations to be encompassed in a list {a=5_m/s^2;vo=5_m/s;...} (or in a vector, if applicable) ? If so, the only practical problem is to access the "=" character while writing a list (or a vector) in the actual key layout of the calculator. That's why I am suggesting that the contextual menu (via the ۞ key) associated to lists (or vectors) might include characters such as "=" (or ";", "◄") to allow a more direct access.

Wiljea, 2024-10-19:
Looking at the actual contextual menu when we have {} or [] on the command line, a possible suggestion is to locate a SpChar sub-menu (for Special Character) in the right corner accessible by Shift Shift F6 giving access to "=", ";", "◄", etc.