c3d/db48x #1271: Compilation error: static declaration of 'sigaction' follows non-static declaration

ulrivo, 2024-10-20:
On a fresh install, doing 'make sim' results in.

recorder.c:1997:20: error: static declaration of 'sigaction' follows non-static declaration
 1997 | static inline void sigaction(int sig, sig_fn *handler, sig_fn *save)
      |                    ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/signal.h:84:5: note: previous declaration is here

   84 | int     sigaction(int, const struct sigaction * __restrict,
      |         ^

c3d, 2024-10-30:
Hi @ulrivo, thanks for sharing.

Could you tell me what is the content of your `config.h`? It's weird that you are in the `!HAVE_SIGACTION` part of the code when you clearly have a `sigaction` definition.

Does that also happen if you directly go to the recorder and to a `git clean -dfx; make`?

Can you share the content of the file `recorder/.logs/make.log`?

ulrivo, 2024-10-30:
Here is my file `recorder/.logs/make.log`:

[make.log](https://github.com/user-attachments/files/17567792/make.log)

c3d, 2024-10-30:
Hi @ulrivo, the Can you share the content of this file: `/Users/ulrivo/dev/db48x/db48x/recorder/.build/macosx-clang/opt/CFGFNsigaction.c".err`?

c3d, 2024-10-30:
Also, what is the version of macOS and the SDK you are using?

ulrivo, 2024-10-30:
macOS Sonoma 14.7

which SDK do you mean? Xcode? Version 16.1

The file `/Users/ulrivo/dev/db48x/db48x/recorder/.build/macosx-clang/opt/CFGFNsigaction.c".err` contains:

`clang: error: unsupported argument 'armv7e-m' to option '-march='
`

c3d, 2024-11-01:
The command line options in `make.log` are indeed quite strange. It looks like a rather unhealthy mix of native and cross compiler

```
-L/usr/local/opt/tcl-tk/lib  -mthumb -march=armv7e-m -mfloat-abi=hard -mfpu=fpv4-sp-d16
```

Does the `/usr/local/opt/tcl-tk/lib` path given to the `-L` option give you any hint? Does it pick that up from your environment?

In any case, if you are compiling on a mac, I would expect the command line to look something like this (one arg per line):

```
gcc
-lm
-lpthread
-framework CoreFoundation
-framework CoreServices
-Wl,-rpath,.
-Wl,-rpath,/Users/ddd/Work/calc/db48x/recorder/
-Wl,-rpath,/usr/local/lib/     	"/Users/ddd/Work/calc/db48x/recorder/.build/macosx-clang/opt/CFGFNsigaction.c" -o "/Users/ddd/Work/calc/db48x/recorder/.build/macosx-clang/opt/CFGFNsigaction.c".exe
```

Your looks like this:

```
gcc
-L/usr/local/opt/tcl-tk/lib
-mthumb
-march=armv7e-m
-mfloat-abi=hard
-mfpu=fpv4-sp-d16
-Tsrc/dm42/stm32_program.ld
-Wl,-Map=build/dm42/release/db48x.map,--cref
-Wl,--gc-sections
-Wl,--wrap=_malloc_r
-lm
-lpthread
-lm
-lpthread
-framework CoreFoundation
-framework CoreServices
-Wl,-rpath,.
-Wl,-rpath,/Users/ulrivo/dev/db48x/db48x/recorder/ -Wl,-rpath,/usr/local/lib/     	"/Users/ulrivo/dev/db48x/db48x/recorder/.build/macosx-clang/opt/CFGFNsigaction.c" -o "/Users/ulrivo/dev/db48x/db48x/recorder/.build/macosx-clang/opt/CFGFNsigaction.c".exe
```

So there is quite a bit of additional stuff.

As far as I can tell, none of this comes from teh `make-it-quick` build system. The most likely explanation is that you have `CFLAGS` or `CXXFLAGS` set that contains these values.

Can you show what `env` says in the environment where you build?

ulrivo, 2024-11-03:
First: thank you very much for your effort to help me. Still not solved :-(

I assume that the `-march=armv7e-m, -mfloat, -mthumb, -mfpu..` options are needed to compile for my Apple Silicon (M1) processor (??)

There were unwanted CPPFLAG and LDFLAGS in my .zshrc
`export CPPFLAGS="-I/usr/local/opt/tcl-tk/include"`
which I removed.

Now I get a different error message:

```
arm-none-eabi-gcc -c -mthumb -march=armv7e-m -mfloat-abi=hard -mfpu=fpv4-sp-d16 -Wno-packed-bitfield-compat -Wall -fdata-sections -ffunction-sections  -DNDEBUG -DDM42 -DHELPFILE_NAME=\"/help/db48x.md\" -DHELPINDEX_NAME=\"/help/db48x.idx\" -D__weak="__attribute__((weak))" -D__packed="__attribute__((__packed__))" -Isrc/dm42 -Isrc/dmcp -Isrc -Idmcp/dmcp -Os -Wno-misleading-indentation  -MD -MP -MF .dep/pgm_syscalls.o.d -Wa,-a,-ad,-alms=build/dm42/release/pgm_syscalls.lst dmcp/dmcp/sys/pgm_syscalls.c -o build/dm42/release/pgm_syscalls.o
dmcp/dmcp/sys/pgm_syscalls.c:44:10: fatal error: sys/stat.h: No such file or directory
   44 | #include <sys/stat.h>
      |          ^~~~~~~~~~~~
compilation terminated.
make: *** [build/dm42/release/pgm_syscalls.o] Error 1
```

ulrivo, 2024-12-02:
I tried to 'make sim' with the actual version 0.8.7 again, and it compiled without any error anymore.  ;-)

closed 2024-12-02 by ulrivo