c3d/db48x #1277: Wrong value of ATAN applied to a ratio of different unit lengths

Wiljea, 2024-10-21:
For instance:  1.5_m  5_cm  ATAN leads to:
ATAN(1.5_m/5_cm) = ATAN(0.3_m/cm) = 1 669.92442 34 ° 
which is obviously wrong because if BASE was performed before, this will give:
ATAN(30) = 88.09084 7567 °

which is correct. Therefore BASE should be performed apriori in such cases.

Wiljea, 2024-10-22:
The same is seen for EXP(-t/(R*C) with C=50_μF R=100_Ω t=2_ms in "RC Transient" where this leads to:
EXP(-¹/₂ ₅₀₀ ms/(μF·Ω)) = 999.60007 9989
which is wrong because, using base
BASE(-¹/₂ ₅₀₀ ms/(μF·Ω)) = -²/₅  =>  EXP(-t/(R*C) = 0.67032 00460 36

c3d, 2024-11-03:
Duplicated by mistake in #1301, which I now closed.

This will be fixed in the next release. Thanks for reporting.

Wiljea, 2024-11-11:
I think that we should close this issue ??

Wiljea, 2024-12-23:
Done.

closed 2024-12-23 by Wiljea