c3d/db48x #1278: "sin error: numerical precision lost"

Wiljea, 2024-10-24:
This occurred in the Library equation with:
t=32_s  Imax=10_A  ω=636_r/s  φ=30_°
@ Expecting [ f=101.22254 3806 Hz 9.59828 06834 1 A ]
@ C#6 NOT OK. MSOLVER: "No solution?", OK solving for f separately BUT "sin error: numerical precision lost" when solving for I_A 
'ROOT(ⒺSinusoidal Current;[f;I];[1_Hz;1_A])'

Then I tried to calculate by hand in the simulator according to the equation:
'(I_A)=(Imax_A)*SIN((ω_(r/s))*(t_s)+(φ_°))'

where the argument (either in RAD or in °) of the SIN function is calculated as:
arg_r=20 352.52359 88 r  OR  arg_°=1 166 113.70465 °  then SIN leads to the error

otherwise, if I calculate without the radian or degree units:
arg_r CONV Value  2·π  MOD = 1.28638 88209 2  MODES Rad SIN => 0.95982 80683 41
arg_° CONV Value 360  MOD = 73.70465 02514  MODES Deg SIN => 0.95982 80683 41

it remains to multiply by (Imax_A) to obtain the expected result:
(10_A)·0.95982 80683 41 = 9.59828 06834 1 A

Therefore there is a problem when calculating SIN of such large arguments.

NOTE. I tried to calculate modulo of two numbers having units:
arg_r  2·π·(1_r)  MOD   OR   arg_° 360°  MOD

and these doesn't evaluate, I had to remove units. For instance, it fails also if try:
'1 000 000 ° mod 360 °'  →NUM

c3d, 2024-10-29:
This is actually a feature, not a bug.

`sin` and similar functions raise this error when the argument is big enough that argument reduction will cause a drop in the number of bits sufficient to render the result somewhat insignificant.

The idea is that I can compute with a few extra digits to avoid rounding errors, but if I want to limit the number of added digits require, I need to limit how many turns I can do. If I add 3 digits, then values above 1000 turns are going to lose precision in the sense that the modulo-reduced angle will not be identical to the original value. At the limit, if you run with 24 digits, a value like 10^25 turns will always reduce to 0.

This is controlled by the `IgnorePrecisionLoss` / `ReportPrecisionLoss` flag.

c3d, 2024-10-29:
Closing as "won't fix", since this is expected behaviour. The `mod` behavior is _not_ expected, but that's a separate  bug.

closed 2024-10-29 by c3d

Wiljea, 2024-10-29:
But in the case of the Library Equation case : "Sinusoidal Current" with:
t=32_s Imax=10_A ω=636_r/s φ=30_°
@ Expecting [ f=101.22254 3806 Hz 9.59828 06834 1 A ]
the mod function applied by hand works since I obtain the expected result which is the same as the HP50g one.
Anyway, because you maintain such limit, I will simply reduce either the time or the ω value of this example. This way the simulation will work. That's ok.