c3d/db48x #1281: MacOS build: wc does not have "-L" option

cdmackay, 2024-10-24:
`make sim` is failing on MacOS 13.7, because its `wc` does not support the `-L` option used.

```
…
cp doc/*.png help/
mkdir -p help/img
grep -b '^#' help/db48x.md | sort -k2 -t: > help/db48x.idx
[ "$(cat help/db48x.idx | wc -L)" -lt 80 ]
wc: illegal option -- L
usage: wc [-clmw] [file ...]
/bin/sh: line 0: [: : integer expression expected
make: *** [help/db48x.idx] Error 2
```

c3d, 2024-10-29:
Are you OK with just removing that line from your `Makefile` if you are running on an old version of macOS?

The intent of this line is to check that the index file entries fit in the internal buffer of the calculator. Any title larger than 80 characters would not fit on the screen of the machine anyway. It is only necessary if you modify the `.md` files used to generate the help (and more specifically if you add new titles). This check matters to me, but it should not matter to someone not doing releases.

cdmackay, 2024-10-30:
Thanks very much. I removed that line, but the make now fails further on, for an unrelated issue:

make: *** No rule to make target `recorder/recorder.h', needed by `recorder/config.h'.  Stop.

I could log an issue for that too. Or do you consider MacOS 13.x too old to support? Unfortunately, my system is constrained to that version, by policy.

cdmackay, 2024-11-08:
Opened new issue #1316