c3d/db48x #1283: Moment of inertia units do not convert from English to SI

Wiljea, 2024-10-25:
I encountered this error condition while working on "Angular Mechanics" of the Equation Library. You could try:
1 750 lb·in↑2  1 kg·m↑2  CONV  Convert
This leads to:
[89621361 52744.773391] errors: Error [Inconsistent units]

c3d, 2024-10-29:
This works for me, and returns `5 121 193 935 011/10 000 000 000 000 kg·m↑2`.

I suspect this is another case of the wrong `*` being in the equation somewhere, but I'd like to know more.

Please copy/paste the exact values from the simulator into this issue.

Marking this issue as `verify`, meaning that I'm awaiting verification that the problem exists.

Wiljea, 2024-11-11:
You were right, so I am closing this issue.

closed 2024-11-11 by Wiljea