c3d/db48x #1286: program with local variables gives error "Malformed local program"

prismv, 2024-10-25:
A program example for 48G/GX with local variables is:

« → x y '2*x+x*y' » 

Stored as q, and put 2 numbers in stack, performing q menu or eval 'q(1,2)' gives 'Malformed local program' error.

c3d, 2024-10-29:
@prismv Thanks for the report. I had trouble figuring out what program you had entered, because the `*` character in Markdown is used to represent italics, as in *italics*. Please use `Code` (i.e. enclose code with the backtick character) to avoid the issue when sharing actual code.

With `« → x y '2*x+x*y' »`, I cannot reproduce the problem. Can you give me more details, including:

* What device you ran this on (simulator, DM32, DM42, etc)?
* What version of the software you ran it on (use the `VERSION` command)?
* How you entered the program (directly, copy/paste, loading a file)?
* In the given configuration, type the following program and share the result:

```
256 STWS 'q' RCL BYTES
```

Thank you in advance.

prismv, 2024-10-30:
Yeah, putting in all these unicode chars sadly retracted me from usual habits.  Issue text is now corrected. You guessed the right content.

I had not checked this on simulator, but I see now that with this program there is no such issue.

DM42 was used. Version previously used was 0.8.1 or 0.8.2, and actually checked on "0.8.3Z".

The program was entered directly. 

The hex output of requested ...BYTES command is: `0779 0178 0102 1206 1404`<sub>16</sub> and number 22 on second line.

c3d, 2024-10-30:
Thanks @prismv. For now, there are a few things that concern me:

* You mention a `0.8.3Z` version number. That would make sense if you built the firmware yourself (e.g. that's the number you could see in the simulator), but not if you downloaded it from the release. Is `0.8.3Z` from your simulator, or is it from your DM42?

* The `0779 0178 0102 1206 1404` value is too short. With `256 STWS`, you should have more than 64 bits in that object output. The object size is correct, and so are the prologue and the bits you shared.

* The `Malformed local program` message indicates that the object evaluating after your lcoals is not a program. But even with a truncated `BYTES` output, I can still tell that the object format is correct.

Let me decode the output of `BYTES` for you, reading from the end.

* `04` is the prefix for a program. This means that `typename` for this object would return `program`.

* `14` is the size of the program in hexadecimal, i.e. 20 bytes (plus the `04` and `14`, that's the `22` output of the `BYTES` command. Looks OK too.

* `06` indicates that your program contains a `locals` block.

* `12` is the size of the locals block

* `02` is the number of local names, two in your case (`x` and `y`).

* `01` is the length of the first local name, `78` is the ASCII code for `x`.

* `01` is the length of the second local name, `79` is the ASCII code for `y`.

* `07` is the type of the inner program object, in this case an expression.

That `07` alone indicate that you should not get the `Malformed local program`. So I'm really puzzled as to what is happening here.

prismv, 2024-10-31:
0.8.3Z is both, version of simulator and DMCP firmware. And yes, locally compiles with

```
> arm-none-eabi-g++ --version
arm-none-eabi-g++ (Arch Repository) 14.1.0
```

I was able to compile with both, `OPT=release` and `OPT=small` (see also #1106). I may try also other options.

The problem does not exist with your release. So it might be a compiler issue. Do you still use `10.3-2021.10` on Mac? Have you tried recent versions on Fedora? (Just because `dnf install arm-none-eabi-gcc arm-none-eabi-gcc-cs-c++ arm-none-eabi-newlib` is given in the build documents.)

c3d, 2024-11-01:
> The problem does not exist with your release. So it might be a compiler
> issue. Do you still use `10.3-2021.10` on Mac? Have you tried recent versions
> on Fedora? (Just because `dnf install arm-none-eabi-gcc
> arm-none-eabi-gcc-cs-c++ arm-none-eabi-newlib` is given in the build
> documents.)

Worrying if it's a compiler issue. Yes, I'm still on that 10.3 release, and no, I have not tried recent versions on Fedora.

c3d, 2024-11-01:
I'm afraid in that case we are reduced to good old `debug_print` debugging to try to figure out what is happening.

But before we go there, can you try to apply the following patch and tell me what shows up on the stack?

```
diff --git a/src/locals.cc b/src/locals.cc
index 9b98487f..c870349a 100644
--- a/src/locals.cc
+++ b/src/locals.cc
@@ -245,6 +245,7 @@ EVAL_BODY(locals)
     program_p prog = p->as_program();
     if (!prog)
     {
+        rt.push(p);
         rt.malformed_local_program_error();
         return ERROR;
     }
```

Do a `BYTES` on the stack result to see what is inside.

c3d, 2024-11-01:
If you want to add a `debug_printf` statement to see what is happening, it takes a row as a first argument, e.g. `4` beow is roughly mid-screen.

```patch
diff --git a/src/locals.cc b/src/locals.cc
index 9b98487f..bbf761db 100644
--- a/src/locals.cc
+++ b/src/locals.cc
@@ -241,10 +241,15 @@ EVAL_BODY(locals)
         p += nlen;
     }

+    debug_printf(4, "Offset %d len %u names %u type %d",
+                 +p - object_p(o), len, names, p->type());
+    debug_wait(-1);
+
     // Defer execution of body to the caller
     program_p prog = p->as_program();
     if (!prog)
     {
+        rt.push(p);
         rt.malformed_local_program_error();
         return ERROR;
     }
```

For me, it shows `Offset 7 len 18 names 2 type 7`. What do you get?

prismv, 2024-11-01:
I used current dev, i.e. ersion string is 0.8.3-20-01a9Z.

If I add only `rt.push(p)` and add 2 arguments, choose Q in menu, these arguments are removed from stack, and I get as direct output `STO` (w/o `BYTES`) on ERROR (with usual error message). Note that I had performed STO Q just before this test (via VAR menu).

If I insert also the `debug_printf`, I get the same output: like you: `Offset 7 len 18 names 2 type 7`, and there is _no_ error message.
The input is not evaluated correctly:

- Input 2 (line 2) and 0 (line 1) or evaluating 'q(2;0)'` sets 4 in line 2 and 0 in line 2.  
- Input 2 (line 2) and 2 (line 1)  or evaluating 'q(2;2)' sets 4 in line 2 and 4 in line 2.  

I had to insert 'q(2;0)'` and not 'q(2,0)'` to get a result, because the latter is converted to 'q(2.)'` and gives an "Too few arguments" error, if line 2 is empty.

c3d, 2024-11-03:
> I used current dev, i.e. ersion string is 0.8.3-20-01a9Z.

OK.

> If I add only `rt.push(p)` and add 2 arguments, choose Q in menu, these
> arguments are removed from stack, and I get as direct output `STO` (w/o
> `BYTES`) on ERROR (with usual error message). Note that I had performed STO Q
> just before this test (via VAR menu).

I do not understand what you mean with "I get as direct output `STO` (w/o
`BYTES`) on ERROR". Do you mean that `STO` shows up on the stack? What does "on ERROR" mean in that context? What does "w/o `BYTES`" mean in that context?

Maybe a screen capture would help me figure out what you mean.

> If I insert also the `debug_printf`, I get the same output: like you: `Offset
> 7 len 18 names 2 type 7`, and there is _no_ error message.  The input is not
> evaluated correctly:

I think that you are saying that the `debug_printf` alone changes the behaviour compared to just adding `rt.push`? That is really weird.

> - Input 2 (line 2) and 0 (line 1) or evaluating 'q(2;0)'` sets 4 in line 2 and 0 in line 2.
> - Input 2 (line 2) and 2 (line 1)  or evaluating 'q(2;2)' sets 4 in line 2 and 4 in line 2.

This is partially incomprehensible to me, because you are repeating "in line 2". I believe that the second "in line 2" on each row is supposed to be "in line 1", and by "line", I believe you mean stack level.

If so, this is really interesting, because it suggests that the reason the program is malformed is because there is a value `4` in the program instead of the expression.

I'm a bit curious if you would happen to have either of:

* `NumericalResults` flag set
* `x` or `y` global variables in the path

> I had to insert `'q(2;0)'` and not `'q(2,0)'` to get a result, because the
> latter is converted to 'q(2.)'` and gives an "Too few arguments" error, if
> line 2 is empty.

Yes, this is intentional. DB48X always considers `,` as a decimal separator and requires a `;` as an argument separators. It encourages you do to so by making it easy to enter `;` using the SPC key when inside parentheses.

prismv, 2024-11-03:
> OK.
>
>
> I do not understand what you mean with "I get as direct output `STO` (w/o `BYTES`) on ERROR". Do you mean that `STO` shows up on the stack? What does "on ERROR" mean in that context? What does [...]

Yes, `STO` shows up on the stack - without applying BYTES command - together with the error message "... Malformed local program" (I should have written "on error").

> Maybe a screen capture would help me figure out what you mean.
>
>
> I think that you are saying that the `debug_printf` alone changes the behaviour compared to just adding `rt.push`? That is really [...]

Yes, the `debug_printf ` or `debug_wait(-1)` statement before `if (!prog)` seems to be the reason that `!prog` evaluates to `false`.

> This is partially incomprehensible to me, because you are repeating "in line 2". I believe that the second "in line 2" on each row is supposed to be "in line 1", and by "line", I believe you mean [...]

line [nr] is the stack line with the number shown on screen. 

I could have written Y for 2, and X for 1 in RPN style, but the assignment to q(x,y) is just the other way around (as seen in simulator): Y or 2 sets x, X or 1 sets y.

> If so, this is really interesting, because it suggests that the reason the program is malformed is because there is a value `4` in the program instead of the expression.
> 

I tested some other examples. I believe that only x is set by stack line 2, and content of 1 is irrelevant, and y is 0.

### ex1

Stack 2: 1
Stack 1: 5
[q]                  <- selected VAR menu entry

or `q(1 5) EVAL` 

gives result

Stack 2: 2
Stack 1: 5

### ex2

Stack 2: -1
Stack 1: 5
[q]

gives result

Stack 2: -2
Stack 1: -5

but 

`q(-1 5) EVAL` 

gives result

Stack 2: 2
Stack 1: -5

### ex 3

Stack 2: -1
Stack 1: 0.1

gives result

Stack 2: -2
Stack 1: -0.1

but

`q(-1 0.1) EVAL` 

gives result

Stack 2: 2
Stack 1: -0.1

> I'm a bit curious if you would happen to have either of:
> 
>     * `NumericalResults` flag set
> 
>     * `x` or `y` global variables in the path
> 
> 

`x` or `y` are only defined in q program.

`'NumericalResults' FS?` returns `FALSE`

> Yes, this is intentional. DB48X always considers `,` as a decimal separator and requires a `;` as an argument separators. It encourages you do to so by making it easy to enter `;` using the SPC key [...]

"SPC": Good to know.

prismv, 2025-04-17:
@c3d GCC 14.2.0 is also not usable for programs that use local variables.

It might be good to mention in documentation the supported GCC version, which is shown on my system as:

```
arm-none-eabi-g++ (GNU Arm Embedded Toolchain 10.3-2021.10) 10.3.1 20210824 (release)
```

Might be that version 12.4 also works fine, but I cannot test it with a simple package install on my Arch PC.

The statement 'or installed directly on many platforms.' in current [BUILD.md](https://github.com/c3d/db48x/blob/dev/BUILD.md) is therefore currently not 'safe',
because newer versions 14.x are nowadays shipped, 
e.g. withFedora >=40 or EPEL 10.1.

be-philippe, 2025-04-21:
For a detailed explanation and a possible workaround, see my latest comments in #1458.

c3d, 2025-04-23:
Removing `verify` tag since this is confirmed by @be-philippe in #1458

c3d, 2025-04-24:
I don't see the problem with the GCC variant on Fedora 41 (arm-none-eabi-gcc (Fedora 14.1.0-2.fc41) 14.1.0).

closed 2025-05-16 by prismv