c3d/db48x #1289: Power & energy units conversion error

Wiljea, 2024-10-28:
This happened in "Conduction" while calculating by hand the following equation with the input data:
Tc=25_°C  Th=75_°C  A=12.5_m^2  L=1.5_cm  k=0.12_W/(m·K)
'(qr_W)=((k_(W/(m*K)))·(A_(m^2))/(L_m))·((Th_°C)-(Tc_°C))'

we obtained:
qr=50. K·m↑2·W/(m∗K·cm) which should simplify to 5000_W

but instead, we had:
50. K·m↑2·W/(m∗K·cm)  1_W  CONV Convert 
[35095210 174918.645019] errors: Error [Inconsistent units]

This might explain why ROOT is unable to solve in "Conduction" & in "Heat Capacity" (where energy units are present) either in the MSOLVE mode or when we tried to solve for each unknown separately.

c3d, 2024-10-29:
You got bitten by Notepad again ;-)

The `∗` character in your example is Unicode 8727, not `*`, Unicode 42.

It is not recognized by DB48X as a multiplication.

c3d, 2024-10-29:
Tested with the normal multiplication, the conversion happens correctly.

I've replaced the `∗` characters with `*` in your latest drop. This is part of the changes that were lost in translation.

Closing this issue.

closed 2024-10-29 by c3d