c3d/db48x #1316: MacOS make sim: No rule to make target `recorder/recorder.h', needed by `recorder/config.h'.

cdmackay, 2024-11-08:
After commenting out the `wc -L` in Makefile, as per Issue #1281 , my make now fails further on.

This on MacOS 13.7.1

[make.txt](https://github.com/user-attachments/files/17682661/make.txt)

bentitmus, 2024-11-09:
You need to check out submodules:

```git submodule update --init --recursive .```

in the root directory.

cdmackay, 2024-11-09:
Thanks Ben! That got a little further:

```
cd recorder && /Library/Developer/CommandLineTools/usr/bin/make TARGET=opt
make-it-quick/rules.mk:367: warning: overriding commands for target `/Volumes/T5'
make-it-quick/rules.mk:365: warning: ignoring old commands for target `/Volumes/T5'
make-it-quick/rules.mk:680: warning: overriding commands for target `/Volumes/T5'
make-it-quick/rules.mk:367: warning: ignoring old commands for target `/Volumes/T5'
[GENERATE]       T5
/bin/bash: /Volumes/T5: Operation not permitted
make[1]: *** [/Volumes/T5] Error 1
make: *** [recorder/config.h] Error 2
mbp2018 $ pwd
/Volumes/T5 CDM/Topics-T5/src/db48x
```

Looks like something doesn't like the space in "T5 CDM"?

bentitmus, 2024-11-09:
I think that's because you're setting `VARIANT` to `T5 CDM` somewhere. Maybe check your `env | grep VARIANT` to see if it is set (unless you deliberately did this). If you didn't deliberately set `VARIANT` but it is set in your `env` then try:

```VARIANT=dm42 make sim```

Alternatively if your `VARIANT` isn't being set then it could be that `MOUNTPOINT` is being set to something. In that case deliberately set the `MOUNTPOINT`:

```MOUNTPOINT=/Volumes/dm42/ make sim```

cdmackay, 2024-11-09:
Thanks again Ben. Nothing like that in my environment:

mbp2018 $ env | grep -i -e vol -e mount -e variant -e t5
OLDPWD=/Volumes/T5 CDM/Topics-T5/src/db48x
mbp2018 $

I'll try those manual settings, and report.

bentitmus, 2024-11-09:
Oh, sorry, I misread your previous commands. I suspect the problem is that there is a space in the name of your disk (`T5 CDM`) and somewhere it is incorrectly escaping this in a `Makefile`.

cdmackay, 2024-11-09:
I'm afraid that made no difference:

```
mbp2018 $ export VARIANT=dm42 MOUNTPOINT=/Volumes/dm42/
mbp2018 $ env | grep -i -e vol -e mount -e variant -e t5
PWD=/Volumes/T5 CDM/Topics-T5/src/db48x
MOUNTPOINT=/Volumes/dm42/
VARIANT=dm42

mbp2018 $ make clean
rm -fR .dep build sim/*.o sim/*/*.o

mbp2018 $ make sim
mkdir -p build/dm42/release
touch build/dm42/release/.exists
echo "#define DB48X_VERSION \"0.8.3Z\"" > build/dm42/release/version-0.8.3Z.h
cp build/dm42/release/version-0.8.3Z.h src/dmcp/version.h
cd sim; qmake db48x.pro -o db48x.mak CONFIG+=release
Project WARNING: Qt requires at least version 13 of the platform SDK, you're building against version 10.15.4. Please upgrade.
rsync -av --delete doc/img/*.bmp help/img/
building file list ... done

sent 2047 bytes  received 20 bytes  4134.00 bytes/sec
total size is 373604  speedup is 180.75
mkdir -p help && \
	cat doc/0-Overview.md doc/1-introduction.md doc/2-Quickstart.md doc/3-Types.md doc/4-ReleaseNotes.md doc/5-Unimplemented.md doc/6-Performance.md doc/calc-help/constants.md doc/calc-help/equations.md doc/calc-help/menus.md doc/commands/angles.md doc/commands/arithmetic.md doc/commands/base.md doc/commands/binary.md doc/commands/bitmaps.md doc/commands/comments.md doc/commands/complex.md doc/commands/composites.md doc/commands/constants.md doc/commands/debug.md doc/commands/dirs.md doc/commands/docol.md doc/commands/docol2.md doc/commands/flags.md doc/commands/fonts.md doc/commands/graphics.md doc/commands/lam.md doc/commands/libdata.md doc/commands/libptr.md doc/commands/lists.md doc/commands/matrix.md doc/commands/menus.md doc/commands/numerics.md doc/commands/objects.md doc/commands/plot.md doc/commands/sdcard.md doc/commands/settings.md doc/commands/solvers.md doc/commands/stack.md doc/commands/statistics.md doc/commands/symbolic.md doc/commands/system.md doc/commands/tags.md doc/commands/text.md doc/commands/transcendentals.md doc/commands/ui.md doc/commands/units.md doc/commands/usb.md doc/commands/user-mode.md help/db48x-images | \
	sed -e '/<!--- DM42 --->/,/<!--- !DM42 --->/s/DM42/KEEP_IT/g' \
	    -e '/<!--- DM.* --->/,/<!--- !DM.* --->/d' \
	    -e '/<!--- KEEP_IT --->/d' \
	    -e '/<!--- !KEEP_IT --->/d' \
	    -e 's/KEEP_IT/DM42/g' \
	    -e 's/DB48X/DB48X/g' \
	    -e 's/db48x.md/db48x.md/g' \
            -e 's/DM42/DM42/g' > help/db48x.md
cat: help/db48x-images: No such file or directory
cp doc/*.png help/
mkdir -p help/img
grep -b '^#' help/db48x.md | sort -k2 -t: > help/db48x.idx
cd recorder && /Library/Developer/CommandLineTools/usr/bin/make TARGET=opt
make-it-quick/rules.mk:367: warning: overriding commands for target `/Volumes/T5'
make-it-quick/rules.mk:365: warning: ignoring old commands for target `/Volumes/T5'
make-it-quick/rules.mk:680: warning: overriding commands for target `/Volumes/T5'
make-it-quick/rules.mk:367: warning: ignoring old commands for target `/Volumes/T5'
[GENERATE]       T5
/bin/bash: /Volumes/T5: Operation not permitted
make[1]: *** [/Volumes/T5] Error 1
make: *** [recorder/config.h] Error 2
mbp2018 $
```

The fact that it seems fixated on the current directory truncated at the first space, suggests that something isn't expecting spaces in directory names?

cdmackay, 2024-11-09:
> Oh, sorry, I misread your previous commands. I suspect the problem is that there is a space in the name of your disk (`T5 CDM`) and somewhere it is incorrectly escaping this in a `Makefile`.

Yes, snap :)

I'll log a new issue for that. thanks again for the help.

closed 2024-11-09 by cdmackay