c3d/db48x #1318: Build fails when current directory path contains a space — MacOS

cdmackay, 2024-11-09:
MacOS 13.7.1

Current dir has a space in the path; looks like something in the make process isn't handling that correctly.

```
mbp2018 $ pwd
/Volumes/T5 CDM/Topics-T5/src/db48x

mbp2018 $ make clean
rm -fR .dep build sim/*.o sim/*/*.o

mbp2018 $ make sim
mkdir -p build/dm42/release
touch build/dm42/release/.exists
echo "#define DB48X_VERSION \"0.8.3Z\"" > build/dm42/release/version-0.8.3Z.h
cp build/dm42/release/version-0.8.3Z.h src/dmcp/version.h
cd sim; qmake db48x.pro -o db48x.mak CONFIG+=release
Project WARNING: Qt requires at least version 13 of the platform SDK, you're building against version 10.15.4. Please upgrade.
rsync -av --delete doc/img/*.bmp help/img/
building file list ... done

sent 2047 bytes  received 20 bytes  4134.00 bytes/sec
total size is 373604  speedup is 180.75
mkdir -p help && \
	cat doc/0-Overview.md doc/1-introduction.md doc/2-Quickstart.md doc/3-Types.md doc/4-ReleaseNotes.md doc/5-Unimplemented.md doc/6-Performance.md doc/calc-help/constants.md doc/calc-help/equations.md doc/calc-help/menus.md doc/commands/angles.md doc/commands/arithmetic.md doc/commands/base.md doc/commands/binary.md doc/commands/bitmaps.md doc/commands/comments.md doc/commands/complex.md doc/commands/composites.md doc/commands/constants.md doc/commands/debug.md doc/commands/dirs.md doc/commands/docol.md doc/commands/docol2.md doc/commands/flags.md doc/commands/fonts.md doc/commands/graphics.md doc/commands/lam.md doc/commands/libdata.md doc/commands/libptr.md doc/commands/lists.md doc/commands/matrix.md doc/commands/menus.md doc/commands/numerics.md doc/commands/objects.md doc/commands/plot.md doc/commands/sdcard.md doc/commands/settings.md doc/commands/solvers.md doc/commands/stack.md doc/commands/statistics.md doc/commands/symbolic.md doc/commands/system.md doc/commands/tags.md doc/commands/text.md doc/commands/transcendentals.md doc/commands/ui.md doc/commands/units.md doc/commands/usb.md doc/commands/user-mode.md help/db48x-images | \
	sed -e '/<!--- DM42 --->/,/<!--- !DM42 --->/s/DM42/KEEP_IT/g' \
	    -e '/<!--- DM.* --->/,/<!--- !DM.* --->/d' \
	    -e '/<!--- KEEP_IT --->/d' \
	    -e '/<!--- !KEEP_IT --->/d' \
	    -e 's/KEEP_IT/DM42/g' \
	    -e 's/DB48X/DB48X/g' \
	    -e 's/db48x.md/db48x.md/g' \
            -e 's/DM42/DM42/g' > help/db48x.md
cat: help/db48x-images: No such file or directory
cp doc/*.png help/
mkdir -p help/img
grep -b '^#' help/db48x.md | sort -k2 -t: > help/db48x.idx
cd recorder && /Library/Developer/CommandLineTools/usr/bin/make TARGET=opt
make-it-quick/rules.mk:367: warning: overriding commands for target `/Volumes/T5'
make-it-quick/rules.mk:365: warning: ignoring old commands for target `/Volumes/T5'
make-it-quick/rules.mk:680: warning: overriding commands for target `/Volumes/T5'
make-it-quick/rules.mk:367: warning: ignoring old commands for target `/Volumes/T5'
[GENERATE]       T5
/bin/bash: /Volumes/T5: Operation not permitted
make[1]: *** [/Volumes/T5] Error 1
make: *** [recorder/config.h] Error 2
```

bentitmus, 2024-11-09:
I think this is actually a bug in the [recorder](https://github.com/c3d/recorder) repo. I think the problem is with the use of `firstword` in lines 365 and 367 of `make-it-quick/rules.mk` which I don't think works when you have a space in the path.

cdmackay, 2024-11-09:
Thanks Ben.

Unfortunately, this is now moot for me, since I get a separate failure when building in a dir path without a space. That new failure is because my xcode-tools are too old. But I cannot upgrade those (reasons), and so I cannot build db48x. sigh.

But the issue is still valid, I think. It's just that fixing it won't help me.