c3d/db48x #1325: Occasional failure of Simple Deflection test in Column and Beams

c3d, 2024-11-11:
The following test occasionally fails

```c++
    step("Solving Simple Deflection")
        .test(CLEAR, ID_EquationsMenu, F2, RSHIFT, F3)
        .test("10_ft", NOSHIFT, F2)
        .test("17_ft", NOSHIFT, F3)
        .test("29000000_psi", NOSHIFT, F4)
        .test("40_in^4", NOSHIFT, F5, F6)
        .test("20_ft", NOSHIFT, F1)
        .test("3687.81_ft*lbf", NOSHIFT, F2)
        .test("674.427_lbf", NOSHIFT, F3)
        .test("102.783_lbf/ft", NOSHIFT, F4)
        .test("9_ft", NOSHIFT, F5)
        .test(F6, LSHIFT, F1)
        .expect("y=-1.52523 29401 2 cm")
        .test("1_in", F1, LSHIFT, F1)
        .expect("y=-0.60048 54094 96 in");

```

It fails at the `1_in` case with an inconsistent unit error message.

c3d, 2024-11-11:
The underlying problem is a cascade of failures as follows:

1. The unit factor `o / u` in `unit::convert` is not a `real` for a conversion that should succeed.

2. The reason is that one fo the unit expressions contains a `symbol` for `in` in a multiplication

3. This is because in `unit::mode` the lookup for `in` failed

4. This happens because the evaluation of `254/10_mm` by `object::parse` returns a really bogus return size.

5. This happens because the length returned by the `integer` parser is `s - p.source`, but `s` has not been adjusted during a garbage collection...

6. The garbage collection happens while building the fraction object for `254/10`.

Fixing by keeping a garbage-collected pointer for `s` while building the fractions.

Wiljea, 2024-11-11:
May be this is related to the frequent occurrences of "Inconsistent units" error in the Equation Library as in ISSUE#1323 ?

commit 0d799d20 (2024-11-11): integer: Preserve source offset correctly for fractions

When we parse a fraction, we need to preserve the value of `s`
correctly in order to correctly compute the parsed size.

This was caught by a case where parsing `254/10_mm` would
garbage-collect during the parsing of `254/10`, causing the `_mm` to
be skipped, which in turn led to bogus inconsistent units message.

Fixes: #1325
closed 2024-11-11 by c3d