c3d/db48x #1341: transient alpha does not work for input 't' and 'P'

prismv, 2024-11-16:
▼+[4] and ▲+[7] does not provide transient input of 't' and 'P', respectively, but output of number and erroneous (cursor?) action indicated with a beep.

(tested with current dev branch c6d06a8e)

c3d, 2024-11-16:
I don't think that's a software problem. I just tested on both DM32 and DM42, works perfectly for me.

However, I used to have the problem with my DM42 (only with `t`, not `p`; I also had a separate problem with the `C` key), and it was just that these keys are the closest to the arrow keys, and pushing the arrow key down is sufficient to make the contact with the nearby key become less reliable.

I ended up opening the calculator, readjusted the domes, and fixed the problesm both for C and T keys.

I hope that this works for you. In any case, I cannot reproduce at all.

raprism, 2024-11-17:
Thanks. Good to hear that it is not a bug. Bad for the hardware. Thank you for the hint to your similar problem and mitigation!

closed 2024-11-17 by prismv

prismv, 2024-11-17:
Just for completeness: It turned out to be a similar issue as explained in https://forum.swissmicros.com/viewtopic.php?f=18&t=3768. Isolating the 2 screws with Tesa, only low pressure of tightning, and on the other hand some tape between PCB and back helped to bring back P with ▲+[7] reliably. t with ▼+[4] still needs some firmer press of the arrow key and does not work all the time.