c3d/db48x #1351: Qt issue: Beep is blocking event

prismv, 2024-11-17:
https://www.hpmuseum.org/forum/thread-20157-post-195776.html#pid195776

- On Arch Linux with Qt 6.8.0
  - when beep is enabled in UI, some error message tries to emit beep, but this blocks UI for 30 seconds
  - 1000 .1 BEEP works fine though

As a note on HiDPI: this works out of the box on my PC with wayland compositor, also scaling with environment variable QT_SCALE_FACTOR. E.g. factor of 1.23 works fine, and 2 is too large for my monitor. Option '-s 1.23' gives worse results compared to setting QT_SCALE_FACTOR.

c3d, 2024-12-09:
The delay is really a Qt bug. I have tried on all my Linux hardware, and only one of them shows that behavior. Due to the way the Qt event loop works (only the main thread can process UI-related events), I can't entirely defer the beep to some other thread, so there is very little I can do to fix this bug. The workaround is to use the "silent beep" (flash mode).

prismv, 2024-12-09:
Good to know. Thanks for your check.

closed 2024-12-09 by prismv