c3d/db48x #1353: Sign problems in the display of simple eqns by SOLVE

Wiljea, 2024-11-17:
This can be seen when SOLVE displays the following eqn taken from "Sound Waves":
    "'(as_(cm/(s^2)))=-((ω_(r/s))/(1_r))^2*(s_cm)' "
The negative sign appears inside the square operation. This is an obvious error because what is depicted and calculated is an eqn with the wrong factor which is (-(ω_(r/s))/(1_r))^2*(s_cm).

The same occurs in  "Simple Harmonic" for the 3rd eqn:
    "'(a_(m/s^2))=-(ω0_(r/s))^2/(1_r)^2*(xm_m)*COS((ω0_(r/s))*(t_s)+(φ_°))' "

Wiljea, 2024-12-02:
In version v0.8.7 it is said:
- Fixed the precedence of the unary - operator, so that -X^2 now parses correctly (it used to be interpretd as (-X)^2)

But I checked the case of "Simple Harmonic" in the Equation Library, and the problem remains both in the calculation of acceleration & in the rendering of the equation:
[image]

Then I checked the cases of "Sound Waves" example 1 & 2 and the same is true.

But strangely, there is incoherence in the following calculations using the data:
sm=1e-5_m  ω=2 513.27412 287_r/s  B=141_kPa  ρ=1.29_kg/m^3   x=2_cm   t=0_s  r=5_m   k=7.59297 31808 8_r/m  φ=2_r
(s_cm)='(sm_cm)*SIN((k_(r/cm))*(x_cm)-(ω_(r/s))*(t_s)+(φ_r))'=8.35879 55288 8⁳⁻⁶ m

(as_(cm/(s^2)))='-((ω_(r/s))/(1_r))^2*(s_cm)'=-3 467.23403 848 cm/s↑2
has the right sign but the wrong value which should be -52.79872 32893 cm/s↑2
and (as_(cm/(s^2)))='-1*((ω_(r/s))/(1_r))^2*(s_cm)'=3 467.23403 848 cm/s↑2
which has both wrong sign & value

c3d, 2024-12-02:
Ouch. The precedence is computed incorrectly whenever there is a parenthesis after the `-`. Fix is on `dev` now.

Wiljea, 2024-12-11:
There is no more problem => closing this issue.

closed 2024-12-11 by Wiljea