c3d/db48x #1356: "Inconsistent units" in a legitimate algebraic expression

Wiljea, 2024-11-19:
When 2 additive terms of an algebraic expression have different units, we get the "Inconsistent units" error message even if they are compatible through CONVERT. For instance in the following sum (1) we find units of electrical field (N/C) and magnetic field (T) (with the relation N/C=T*m/s):

Ey=200_(N/C)   Bz=350_T  β=0.995  γ=10.01252  to be used in the sum
(1)   'γ*((Bz_T)-β/Ⓒc*(Ey_(N/C)))'  followed by "=" gives
 10.01252  350_T  0.00000 06637 93 s·N/(m·C)   and the error message

If you do 1_T  CONVERT => 0.00000 06637 93 T then you terminate by  - * to obtain the legitimate answer: 3 504.38321 361 T

I am not sure but I suspect that such problem might explain the frequent occurrences of this error message when solving in the Equations Library.

prismv, 2024-11-22:
Solving or EvalEq of `'Q=√((L_H)/(C_F))/(R_Ω)'` also fails due to inconsistent units. Not sure, if an explicit `Q_1` might help here.

prismv, 2024-11-22:
Indeed `'ROOT(Q=UBASE EQur;[Q];[1])'` (with `EQur` is the right side of equation) works, but without `UBASE` it does not work. Therefore it seems to be another example for the issue you report here.

commit 068cbf71 (2024-12-01): unit: Simplify units that convert to real numbers

During addition and subtraction, when the second argument is a unit
that can be simplified to a real number, eliminate the unit if
possible and perform the computation with the simplified value.

Fixes: #1356
commit c5271a33 (2024-12-02): unit: Simplify units that convert to real numbers

During addition and subtraction, when the second argument is a unit
that can be simplified to a real number, eliminate the unit if
possible and perform the computation with the simplified value.

Fixes: #1356
closed 2024-12-02 by c3d