c3d/db48x #1357: Heavy battery drain?

neutralinsomniac, 2024-11-20:
I'm on the DM42, and my battery voltage has dropped to 2.7V within 3 days of very light usage of db48x. Is this expected?

c3d, 2024-11-22:
It is relatively expected, in particular if you used a no-brand battery.

Can you try comparing with the same brand and running the reguar DM42 firmware?

neutralinsomniac, 2024-11-22:
Yes, I flashed back to stock firmware and the voltage settled back to around 3.0V. Switched back to db48x, and battery sagged back to 2.7V.

c3d, 2024-12-02:
I see no real reason why DB48x would report a different battery voltage than the stock firmware, except for the fact that maybe it reads it more often? I'm using the exact same function, and when the calculator is idle, it's really doing nothing, so the battery voltage when dawdling should be exactly the same.

I will be experimenting with this to see exactly how the battery evolves over time. One possibility is that I have been a bit too aggressive in setting the limits for battery states.

c3d, 2024-12-02:
So the experiment that is currently running on my DM42, which has a flaky battery (about 2.52V) is as follows: I use the `Header` command to extract the current time as well as the current battery voltage every 10s. I then use the `Σ+` command to collect that information over time, and then the `ScatterPlot` command to draw it. We will see what I get.

While I was experimentally setting the lowbat threshold around 2.5V, I am currently running at 2.436, and it's still not triggering the DMCP builtin low-battery indicator.

c3d, 2024-12-02:
It kept executing well into the 2.2V, but then when I switched the calculator off and then back on, I lost my precious state, which is a bit annoying. I lost all the collected stats data because I was not careful enough to store it to disk every time.

Fortunately, the stats functions are designed for that purpose, and you can put a file name in the `ΣData` variable. That way, even if the calculator crashes, changes are I will still have my data. And it will persist across runs.

c3d, 2024-12-04:
I've reworked the power management a little during the past two days, and experimented with a DM42 containing an almost empty battery, which I use to test low-power situations.

Among other things, I reduced menu animations while on battery power. But otherwise, power management seems to be quite efficient already. I added a new `RuntimeStatistics` command that returns measurements of the time spent running and sleeping. Right now, after sitting next to me for a few minutes, my calculator shows 159640ms sleeping, 4180ms running, and woke up from sleep 140 times. So this means that the animations are not killing the battery, the battery drains while sleeping when the machine is on.

I'm doing my testing using an old battery that presently shows 2.42V right after I unplug from USB. I added some commands providing `BatteryVoltage`, `USBConnected` and `LowBattery`. A new setting, `MinimumBatteryVoltage`, lets you set when the calculator switches off. I've selected a default of 2400 for now, and I'm experimenting with my old battery to check if that value is OK. To me, OK is defined as "the calculator will switch back on without resetting", which appears to be the primary problem I run into when the battery is really low.

Experimentally, I can run the calculator with power voltage showing as 2.1V, but somewhere between 2.1 and 2.4, flash operations seem to start failing, and more importantly, the calculator starts resetting when you switch it back on (but never while running, strangely enough), which is bad because it means that you lose the current state.

I added the `MinimumBatteryVoltage` setting because it seems like the `LowBattery` condition, which returns the DMCP `get_lowbat_state()` value, is not reliable enough. I've observed time and time again that `get_lowbat_state()` is false while the calculator has low-enough battery to fail to switch back on. I'll ask SwissMicros if that is an issue with my DM42 or if it's a known problem.

c3d, 2024-12-04:
An important recommendation is that if the calculator switches off and tells you to connect USB Power, _you should_. If you don't, you may lose the memory state when you switch the calculator on. This happens _even when_ the calculator battery as recovered well enough to load the Demo state from the flash disk and return to the prompt. In other words, the calculator has enough juice to to a whole reset and reload state sequence, but somehow not enough to just switch back on??? That is very weird.

c3d, 2024-12-04:
After a lot of experimentation, I observed that there is a scenario where the internal timer used to detect key repeat would fire up repeatedly, keeping the calculator busy instead of sleeping.

This is fixed now on my `dev` branch.

c3d, 2024-12-09:
Hi @neutralinsomniac. Thanks for drawing my attention to this topic. This ended up being the main focus of 0.8.8, which contains many improvements to reduce battery usage. Please give it a try.

If you load Demo.48S, in the custom menu (which you get with VAR VAR), there is a VStat button that records battery usage statistics in a file called BAT.CSV on your calculator. If you are willing, I would appreciate if you recorded this on a semi-regular interval and shared it on this post, to get data about actual battery voltage over time.

I'm closing the issue for now, feel free to reopen if necessary.

closed 2024-12-09 by c3d