c3d/db48x #1358: Dispatching and assembling tagged matrix objects and lists does not work correctly

skai111, 2024-11-20:
DB50X v.0.8.6-10 -efa5Z
Hardware DM42n

Functions on matrixes which take two objects give incorrect result when first or both matrix are tagged.
e.g 
 1-d matrix or vectors
[5 7] [13 17] * --> [ 65 119 ] // correct
3:[5 7]  11:[13 17] * -->  [[65 85][91 119] // incorrect

matrix
[[ 11 12][21 22]]  DUP * --> [[ 373 396][693 736]] // correct
5:[[11 12][21 22]] DUP * -->
[[[[ 121 132] 
[231 242]]
[[132 144]
[252 264 ]]]
[[[ 231 252]
[441 462]]
[[242 264]
[462 484]]]]   // incorrect 

list 
{ 2 3 } { 5 11 } +  --> { 2 3 5 11 } // correct
23:{ 2 3 } 511:{ 5 11 } + --> { 2 3 511:{5 11} } // inconsistent
comment:
A) The HP48GX strips all tags, then concatenates the lists.
B)  If you want to keep tags, then you can build the new list with 2 -->LIST

c3d, 2024-11-22:
The reports on matrices are valid.

However:

> list { 2 3 } { 5 11 } + --> { 2 3 5 11 }
> // correct 23:{ 2 3 } 511:{ 5 11 } +
> --> { 2 3 511:{5 11} } // inconsistent
> comment:
> A) The HP48GX strips all tags, then concatenates the lists.
> B) If you want to keep tags, then you can build the new list with 2 -->LIST

Actually, it's interesting that HP48SX and HP50G both seem to only strip *both* tags if the first object is a tagged list, but behave like DB48x if second object only is tagged.

``` rpl
{ 1 2 } :A:3 +          --> { 1 2 :A:3 }
{ 1 2 } :A:{3 4} +      --> { 1 2 :A:{ 3 4 } }
```

BUT

``` rpl
:A:{ 1 2 } :A:3 +       --> { 1 2 3 }
:A:{ 1 2 } :A:{3 4} +   --> { 1 2 3 4 }
```

That does not strike me as being particularly consistent and worth replicating.

I think that a more consistent behaviour would be to strip all tags in arithmetic operations, i.e.

``` rpl
{ 1 2 } :A:3 +          --> { 1 2 3 }
{ 1 2 } :A:{3 4} +      --> { 1 2 3 4 }
```

skai111, 2024-11-22:
I think I may have been a bit hasty.
Considering the meaning of +, in DB48x  what should be the action?
For lists, keep it reserved to to build lists?
What about lists inside lists? 
Or possibly a collection of arbitrary objects? 
Why is an object tagged?
 
If one of the arguments is a  list, prepend or append this ist.
If both arguments are lists, explode both and put together again.

if both objects are tagged lists, which list to modify? 
I can think of 3 alternatives:
a)  strip both tags and append them
b)  creating a new empty list and appending both
c) error
Option c should be avoided
Option b looks nice, but is inconsistent with how elements are handled...  ,  Special handling  when both objects have tags to check if they are lists.
Option a is inconsistent with tagged list handling in general.
Also :a:{ 1 2}  :b:{3} :c:{4 5} +  + -->  {:a:{1 2}  3 4 5}  or something similar unexpected for the user ( which happends on the HP48GX)

pest eller kolera (plague or cholera) like we say in norwegian.
Recalling how I have used lists in the past, I think I would prefer option b  and it is consistent with list building using the stack with up-arrow and -->List

I find tagged lists inside list collections to be useful, but mostly as results from a userprogram and temporary storage of data on stack. I am not sure if I actually have built them manually.

On the other hand, to turn the question around, what creates most surprise and irritation fram a user point of view?

Maybe this is as good as it can be?
e.g
Assuming rules: 
1) Any tagged object which is modified must have its tag deleted
2) When appending to a list, keep tags whenever possible to build a collection of arbitrary objects 
3) non-tagged lists are exploded one level and collected

But it does not cover what to do with two tagged objects. HP calculators simply strips both.

PS I am not sure if there is any benefit in keeping or allowing tags inside arrays collections, (vectors, or matrix or mix), it slow everything down.

commit ae12e7ec (2024-11-28): arithmetic: Strip tags on all arithmetic operations

Strip tags in all arithmetic operations, including on arrays/lists.

This introduces a small incompatibility with HP. On HP calculators,
`:A:{1 2} :B:{3 4}` gives `{ 1 2 3 4 }`, but
`{1 2} :B:{3 4}` gives `{ 1 2 :B:{ 3 4 }}`.

This is inconsistent, complicated and causes operations on tagged
arrays to misbehave. Stripping tags in all cases makes things much
more consistent and simpler. This means that adding a tagged object to
a list requires the `→List` operation.

Fixes: #1358
closed 2024-12-02 by c3d