c3d/db48x #1361: Directory not properly initialized when not on first screen page

prismv, 2024-11-23:
If a directory is shown on second screen of HOME, i.e. navigating with arrow (F6) is necessary, then the variables in this directory are at first not available from VariablesMenu.

Only after exiting and reentry of VariablesMenu the variables can be evaluated, read or written by menu actions.

See `test` directory in [RA_00.48S.txt](https://github.com/user-attachments/files/17881629/RA_00.48S.txt)

If a directory is shown on first screen of HOME, this issue does not apply.

c3d, 2024-12-02:
Yes, it is necessary to restore the first page in the menu when changing directories.

c3d, 2024-12-02:
Fix implemented on the `dev` branch. This will be fixed in the next release.

prismv, 2024-12-09:
Checked and okay! Thanks.

commit 66a981ac (2024-12-02): menus: Return to page0 when entering a directory

When we enter a new directory, we need to reset to page 0 before
updating the menu in order to ensure that we generate the correct
commands for the submenus.

Fixes: #1361
closed 2024-12-09 by prismv