c3d/db48x #1364: IOS Key Layout for DB40x

ipxfer, 2024-11-24:
IOS version shows the top row key layout with the wrong label - e.g. y^x should be e^x, etc.  This may be the same for others - DB48x and DB50x.
 
Took forever to figure out why y^x is giving wrong answers.

c3d, 2024-11-30:
This is a known issue, which should be solved in the next release.

uja-works, 2025-02-16:
It seems it didn't make it to last week's release 0.9.0? Same problem here (DB50x on iOS).

LdBeth, 2025-03-02:
Use "Yellow" 0 (SETUP) and `Settings > Load keymap` to pick either `legacy.48k` or `db48x.48k` as the keymap. Also turn on "Save sate when deactivating" to avoid it changes when the app hung up.

orome, 2025-03-02:
That works more often now, but not always.

c3d, 2025-04-28:
There was a race condition between the `application:didFinishLaunchingWithOptions:` notification and the creation of the `ViewController`. This should be fixed in `i0.9.6`, submitted to Apple for review.