c3d/db48x #1370: Unexpected behaviour of y^x function

nickapos, 2024-12-03:
Not sure if I am doing anything wrong but when I am trying to calculate 10^2 using y^x I am getting 
[image]

c3d, 2024-12-04:
This is a know issue with iOS, same as #1364. This should be fixed in the next iOS release.

The problem is that the application displays the wrong keyboard image. It should show [the new layout](https://github.com/c3d/db48x/blob/stable/sim/keyboard-db48x.png), but instead shows the [compatible one](https://github.com/c3d/db48x/blob/stable/sim/keyboard-db48x-old.png).

A workaround for now is to use the SETUP command (Shift 0), then option 1 - Settings, then option 6 - Load keymap, and select the [db48x.48k](https://github.com/c3d/db48x/blob/stable/config/db48x.48k) file.

Uli-Moin, 2025-01-29:
> This is a know issue with iOS, same as [#1364](https://github.com/c3d/db48x/issues/1364). This should be fixed in the next iOS release.
>
> The problem is that the application displays the wrong [...]

The workaround helps, thank you.  Unfortunately I have to load the keymap file each time after start the calculator again.