c3d/db48x #1373: Parsing error of negative constants

Wiljea, 2024-12-04:
For instance the neutron magnetic moment (like 3 other constants) has a negative value. The usual way to specify it in the "Constants.cc" file would be:
"μn",       "−9.6623653E-27_J/T"  but this  leads to  '→NUM(Ⓒμn)'=−9 
which is not even a number.

The only way that I found to circumvent this problem is to write the operation:
"μn",       "'(-1)*9.6623653E-27_J/T'" => '→NUM(Ⓒμn)'=-9.66236 53000 0000⁳⁻²⁷ J/T
which is what was intended

c3d, 2024-12-04:
I am afraid that we have Notepad to blame again ;-)

I copied your code:

``` c++
"μn", "−9.6623653E-27_J/T"
```

and indeed, this does not work. What you get on the stack is `−9`. If you look carefully, you will noticed that it shows as _italics_. This is a hint. It indicates that we have a _symbol_, not a number. And indeed, if we check with `TypeName`, the result is `"expression"`, not `"decimal"` or `"neg_integer"`.

So what is going on?

The first character in your expression is `−` (Unicode 8722) and not `-` (Unicode 45). So the parser creates a name with it. It's `−9` much like you would have `A9`.

I will add an exclusion rule so that the characters that Notepad keeps helpfully generating for you are no longer silently accepted. They will now flat-out be rejected with a syntax error.

Wiljea, 2024-12-04:
It must be a problem of working in the Win10 environment because not only I didn't use NotePad++ but I was copying and pasting strictly between the open file (in edition mode) constants.cc (on GitHub) directly to the simulator via <CCTRL>C, <CTRL>V. I am not working in NotePad++ anymore since I am doing Commits as often as needed.

commit 47053a85 (2024-12-07): parser: Reject Notepad-generated `-` and `*` signs

For some reason, Notepad thinks it's smart to replace the one-byte `-`
`/` and `*` with 3-bytes equivalents `−`, `∕` and `∗`.

Report a syntax error if such characters are encountered by explicitly
rejecting them from names, instead of generating bogus names such as
`−9` or `A∗3`.

Fixes: #1373
closed 2024-12-10 by c3d