c3d/db48x #1377: Possible correction to check_qspi_crc in db48x/tools for Fedora WSL?

msrgm, 2024-12-06:
In my Fedora WSL distribution (distro 40) the 32-bit CRC command is 'cksum', not 'crc32'.  
This is important because the script check_qspi_crc in the release of db48x refers to crc32.

I have edited my copy of the check_qspi_crc script to reflect this, in order for the script and the build to succeed.
(Maybe I have missed something? ) 

The script in the build of 0.8.7 is here: https://github.com/c3d/db48x/blob/stable/tools/check_qspi_crc
The line I have changed locally is line 27:

From this:
ncrc=0x`crc32 $q`

to this:
ncrc=0x`cksum $q`

msrgm, 2024-12-07:
OK, so it seems the checksum should be run be forcecrc32 and not the "native" cksum in Fedora.

msrgm, 2024-12-07:
Closing because it is not an issue.

prismv, 2024-12-08:
It _is_ an issue, because the dependency on `crc32` program is not documented. E.g. "perl-Archive-Zip" for Fedora.

msrgm, 2024-12-09:
Hi prismv, I have raised it with Christophe and he will update the installation documentation.

prismv, 2024-12-09:
> [Verfolgungsbild][https://github.com/notifications/beacon/ACLSOQAE7S53EBDRMF7PGH32EVV3DA5CNFSM6AAAAABTFANEFKWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTUWU3N2Y.gif]

Hi Mark,

great that docs will be improved.

Best regards
Ralf