c3d/db48x #1381: No DB50X folder on iOS following 0.8.8 update

orome, 2024-12-09:
After updating to 0.0.8 on iOS the DB50X folder is gone (so no way to save or load state, or set keymaps). This happens after a normal iOS I update and after a remove-reinstall.

orome, 2024-12-09:
Rebooting the iOS device and then launching the app restores the folder, however it is then still necessary to manually load the [new keymap](
[image]
) in order to resolve #1375 and #1364.

(On macOS the DB50X folder is still there but the keymap only updates if loaded manually or state is loaded from disk.)

c3d, 2024-12-10:
Thanks for the report, @orome

I have observed the "directory disappears until reboot" a couple of times on my own iPhone during development, as well as on iOS simulator, and frankly I have no explanation for it at the moment.

This is certainly related to how iOS implements application sandboxing, but I don't understand why an update of the same application would hide the directory created by the application until you reboot. And if that behaviour is somehow intentional, why is it not happening every time I update the application during development?

c3d, 2024-12-10:
Regarding the comment on "manually load the new keymap", you link to the old one. So just to clarify: the new layout is where you see the MTH menu and have `y^x` as a primary, not secondary function.

If you want the old key layout, then you can load it, but that's a user choice. Since that user preference is stored in the DB50x directory, if that directory disappear, then it's only logical that the preference stored there would as well.

This is a case where I could store that using the iOS support for preferences, the same way I do for iOS-specific settings like haptic feedback. But the intent was to use the same code as on the actual calculator for any preference that exists on the calculator (e.g. display mode, stack content, whatever). So since the calculator can switch keyboard layout, the shared code is used for that, and relies on the filesystem being available.

orome, 2024-12-11:
> Regarding the comment on "manually load the new keymap", you link to the old one. So just to clarify: the new layout is where you see the MTH menu and have `y^x` as a primary, not secondary function.

Yes. sorry: I meant the one with **MTH** as primary. That's what I need to load manually after updating.
On subsequent uses the calculator loads with the old keycap, but switches to the new one wien turned on.

> If you want the old key layout, then you can load it, but that's a user choice. Since that user preference is stored in the DB50x directory, if that directory disappear, then it's only logical that the preference stored there would as well.

Where is that preference stored?

> ...the intent was to use the same code as on the actual calculator for any preference that exists on the calculator.

I think that's the way to go. Handling anhthgn related directly to the actual calculator differently on macOS vs iOS vs DM42 seems like unnecessary work.

c3d, 2025-11-08:
I believe that this issue is now fixed. Please reopen if I'm wrong.