c3d/db48x #1383: Const. π not always saved properly (and: DERIVATIVEx broken?)

prismv, 2024-12-10:
v0.8.8, self-compiled, test in simulator only

On save the following state file is changed with `Ⓒπ` exchanged in "deriv" directory with some other constant (constant in "perf" directory remains the same). Message in terminal is `file_error: open is opening 2 files at the same time ([...]/db48x/state/RA_test.48S and config/constants.csv)`.

Furthermore DERIVATIVEx was evaluating properly before, but not with current version.

```
Directory {
	deriv
		Directory {
			dfval 'f′(x)|x=Ⓒπ÷4'
			f′ 'DERIVATIVEx(f(x))'
			f 'x²+x·sin x'
		}
	perf
		Directory {
			Speed
				«
					Ⓒπ →Num 6.30
					÷
					«
						1 10000
						start
							Duplicate sin Drop
						next
					»
					TimedEval
				»
		}
}
'test' STO

UpperCase
VariablesMenu
 HomeDirectory test deriv

```

c3d, 2024-12-10:
Two bugs here. Thanks for reporting, @prismv, but please file one issue per bug in the future, it will help me track them ;-). Let me do the splitting this time.

Bug 1: Two open files for constants, which I will use in this one.

BUg 2: Wrong spelling for `∂x`, which I filed as #1384.

c3d, 2024-12-10:
The problem with two open files is something I also saw in a private email sent to me by @wiljea. So thanks for filing the issue, @prismv, that saves me from doing so myself ;-)

c3d, 2024-12-10:
There is code in place that is supposed to deal with this by testing for `r.file_save()`:

``` c++
size_t constant::do_rendering(config_r cfg, constant_p o, renderer &r)
// ----------------------------------------------------------------------------
//   Rendering of a constant
// ----------------------------------------------------------------------------
{
    constant_g cst = o;
    size_t     len = 0;
    utf8       txt = nullptr;
    if (file *saving = r.file_save())
    {
        // Saving: can't have two files open at once on DMCP
        file_closer_while_writing fc(*saving);
        txt = cst->do_name(cfg, &len);
    }
    else
    {
        txt = cst->do_name(cfg, &len);
    }
    if (r.editing())
        r.put(cfg.prefix);
    r.put(txt, len);
    return r.size();
}
```

However, by the time we reach this point, the `r.saving` field is null.

c3d, 2024-12-10:
The problem is caused by expression rendering using `as_symbol` directly, which creates a text renderer, ignoring if a file is already open.