c3d/db48x #1384: DERIVATIVE rendering is wrong when saving on disk

c3d, 2024-12-10:
This is lifted from issue #1383 

State file containing `∂x` is written as `DERIVATIVEx`, which does not parse back correctly.

Example given in #1383 by @prismv:

```rpl
Directory {
	deriv
		Directory {
			dfval 'f′(x)|x=Ⓒπ÷4'
			f′ 'DERIVATIVEx(f(x))'
			f 'x²+x·sin x'
		}
	perf
		Directory {
			Speed
				«
					Ⓒπ →Num 6.30
					÷
					«
						1 10000
						start
							Duplicate sin Drop
						next
					»
					TimedEval
				»
		}
}
'test' STO
UpperCase
VariablesMenu
 HomeDirectory test deriv
```

I suspect that the `UpperCase` setting matters to this test case.

c3d, 2024-12-10:
The spelling `DERIVATIVEx` on output is the problem.

It's supposed to be spelled `∂x` and is parsed specially, so if it's written in the state file as `DERIVATIVEx`, then the special parsing does not work.

prismv, 2024-12-11:
Fixed. Could be closed (I cannot here)

commit b51247e4 (2024-12-10): expression: Fix rendering of `derivative`, `primitive` and `where`

In compatibility mode, `∂` renders as something like `Derivative`.
However, this should not happen inside an expression, where we want to
have `∂x(x+1)` and not `DERIVATIVEx(x+1)`.

Fixes: #1384
closed 2024-12-30 by c3d