c3d/db48x #1387: Keymap not carried over in save state

Amarthol78, 2024-12-11:
I changed my keymap to the legacy one, then saved the calculator state. I later recalled the state but the default keymap was restored. I then had to go back and change the keymap back.

c3d, 2024-12-11:
Unfortunately, your description lacks a few details, so I am not sure if this is a bug or the way it's supposed to work.

The keymap is not part of the state file. This is intentional. Like anything under "config" or "library" (e.g. custom units, custom character menus or custom library entries), this belongs to user-customizable aspects of the calculator that are shared across all state files.

The actual file being used when you restart the calculator is called config/keymap.cfg, and it simply contains the name of the keymap file to load.
So if you want this to be preserved across updates, you need to restore your version after updating.

This is part of another deliberate strategy with respect to distributions, which will change after 1.0. Before 1.0, there is no guarantee that configuration file formats, file names or any other element of the calculator state will be compatible from one release to the next. So the distribution overwrites everything, including `config/keymap.cfg` or `config/units.csv` for example. If you made change to these compared to the base distribution, the changes will be overwritten.

Before we reach 1.0, a different distribution mechanism will be put in place, with default files that will be used to initialize configuration files only if they don't exist. This will happen when it seems like configuration files have become sufficiently stable.

prismv, 2024-12-12:
I can confirm this issue on DM42:

1.  config file keymap.cfg contains: `/config/42style.48k`, local state file used
2. state file changed to e.g. Demo.48s
3. Check, if input of 10 and standard [LOG] key (position D) works - No.  Bad argument error.
   Pressing D position longer gives help for `Y^X`
4. Check, if keymap.cfg was changed - No. Still contains `/config/42style.48k`
5. OFF and ON does not change anything
6. User has to manually enable again Setup->Settings->Load keymap->42style.48k 

Conclusion: After state file content of keymap.cfg is not properly evaluated. It is also not evaluated after OFF ON toggle.

Amarthol78, 2024-12-12:
Ok, I have an overlay, so to use it I loaded the legacy keymap. I saved the calculator state and then played around with the calculator.  I then loaded the state that I saved earlier, but the keymap no longer matched my overlay. So I just figured that if I deliberately changed the keymap that my change would be saved in the save state file.

prismv, 2024-12-13:
With some more testing I found out with the simulator set to 42style, that the key assigments are changed also on simulator, _but only_ if I load the currently _distributed Demo.48S_ state file. Other (all significantly shorter) state files I have do not cause this issue on simulator.

Then I was trying to find out, what part of Demo state file may cause the issue, but then I saw that already the updated state file (e.g. the version generated at shutdown of simulator) does not cause this issue anymore.

On hardware (DM42) this issue happens always with Demo.48S, also a newly saved version, but I recognized that the other (smaller, <=4k) state files I have also do _not_ trigger this issue.

Test was always:

- set 42style keymap
- change state file
- test, if LOG (on D key) works as expected
- failed case was always that D key is Y^X

c3d, 2024-12-30:
Confirmed the bug now that I have a more concise description. Sorry for not having seen that issue before releasing 0.8.9. The fix will be in 0.8.10.

prismv, 2024-12-30:
Commit 2f28355 fixes issue for me. Thanks.

commit 2f283550 (2024-12-30): ui: Reload keymap after loading state file

When we load a state file, we reset the memory state before loading.
This clears the keymap among other things. Therefore, reload the
keymap file after resetting memory.

Fixes: #1387
closed 2025-01-13 by c3d