c3d/db48x #1388: In Equations Library: impossibility to import RPL code from Help File to the stack

Wiljea, 2024-12-11:
Simulator DB48X 0.8.8-187-788e3, this functionality was working in previous version. Now if I open the help file (long press on "Columns and Beams") and scroll to the RPL code of "Elastic Buckling" and hit ENTER, I get the following error message "Help error: No help for CantileverShear" as this can be seen:
[image]

This message seems incoherent with the fact that I am actually not viewing the referred section "CantileverShear".

But strangely if I scroll to the next section "Eccentric Columns", everything works fine. Let me check if the bug is reproducible. Yes it is.

HOWEVER if, beforehand, I choose the main section "Columns and Beams" and I call then (long pressing) the help for the subsection "Eccentric Columns", everything works fine.

Let me see, in the case of the next section "Electricity". Scrolling then to the subsection "Coulomb’s Law & E Field" poses no problem. It works fine.

A last check is to restart all-over again with the section "Columns and Beams": same problem.

Wiljea, 2024-12-11:
In the subsection "E Field Infinite Line", a different instance of the same type of problem is when I try to copy to the stack the simple substitution RPL code:
[image]

Wiljea, 2024-12-11:
In the case of "E Field Infinite Plate", pointing directly in this subsection, the RPL code doesn't paste either:
[image]

Therefore, since I am in the process of recalculating the exact answers, here I have to resort to copy-paste from the file "Equations.md".

The problem being generic, I will stop to report them here.

c3d, 2024-12-14:
The problem appears restricted to the `eqns` branch, it does not occur on `dev`. Bisecting.

c3d, 2024-12-14:
Bisecting is complicated by the presence of versions that do not compile, e.g. `[233f74539730470daa9e5e70de6a5a1a74ecd429] Update equations_JW_24_11_12_AM.cc` which has a stray `zzz` inside.

c3d, 2024-12-14:
Also annoying that the commits were made without a build, so that means the build creates the `db48x.md` file, causing `git bisect` to fail at the next iteration because it would "overwrite the changes".

closed 2026-06-16 by Wiljea