c3d/db48x #1392: Combinations with hwfp does not return Inf, if applicable

prismv, 2024-12-12:
When HwFP is enabled, entering 2E929 gives infinity (as expected).

9999 555 COMB would return a decimal result of the same order of magnitude as the previous input, i.e. 2.24471E929, but the operation fails silently - as if COMB had not been pressed - if HwFP is set.

c3d, 2024-12-12:
This does not match my experience. For `9999 555 COMB`, I get `∞` with `hwdouble` or `hwfloat` types with precision 16 and 7 respectively, with an error message `Number is too big`. Just tested on the `dev` branch.

On the other hand, `9999. 555. COMB`, i.e. using decimal dot, gives the wrong result `1`. This is due to the symbolic simplification `X/X=1` applying to the case where `X` is `Inf`. The correct result should be `NaN` for the division. But that does not match your description either, i.e. it does not fail silently.

c3d, 2024-12-12:
Can you copy here the result of the `MODES` command?

prismv, 2024-12-12:
About shows actual HEAD hash: v0.8.8-9-a1f5Z

MODES gave:

```
«
	NUMERICALRESULTS NOAUTOSIMPLIFY
	COMPLEXRESULTS OVERFLOWINDICATOR
	BEEPOFF SILENTBEEPON
	INDEXWRAPPED HARDWAREFLOATINGPOINT
	5 SIG 5 DISPLAYDIGITS
	5 STANDARDEXPONENT 16
	PRECISION UPPERCASE
»
```

prismv, 2024-12-14:
Tested this with 5572b7a2 as successor of 3a0a1c04, but `9999 555 COMB` still fails silently!

skai111, 2024-12-15:
This may be due to overflow of intermediate results, or possibly that the standard for HwFP "gracefully" extend limited precision at boundaries. Consider following examples with single precision.
A1)  set HwFP 5 Prec -100 exp  // natural antilogaritm
result 3.784e-44
A2) deselect HwFP 5 Prec -100 exp
result 3.722e-44

B1) set HwFP 5 prec
B2)  [[1e19 1e19 ][1e19  1e19]] -->NUM // 26byte array of HwFP
B3) DET 
Result 0 
Which is wrong, due to
 aa) intermediate overflow
 bb) wrong handling of infinity - infinity

Similar with double precision, but larger numbers of course.

commit 3a0a1c04 (2024-12-13): simplifications: Do not simplify infinities

When operations produce infinities, make sure we do not apply the
usual simplification rules, e.g. `x/x=1`.

This is particularly important to avoid absurd results from operations
that can overflow and could otherwise return `1` for `COMB` with large
input values.

Fixes: #1392
closed 2024-12-18 by c3d