c3d/db48x #1394: Issue with  HwFP setting and parsing of programs

skai111, 2024-12-15:
DB50X  v. 0.8.8
HW DM42n 
 Consider the below program  with HwFP set.
Manually set HwFP 5 Prec
Enter the following short program:

《  1E11 》
edit the program  (using arrow-down)
《 9.9999997952E10 》
 
Not sure if this is an issue as  the precision is ok.
However, it does affect parsing of programs, so perhaps a warning is reasonable?
It is suggested to amend the helpfile with more explanation.

prismv, 2024-12-15:
Precision of 5 is like setting 7 and is type "hwfloat" (32 bit), above 7 up to 16 is "hwdouble" (64 bit)

It's not the parsing, but already the creation that uses the set floating point type. It is also possible to enter:

```
HardFP
16 PREC
.2
17 PREC
.2
-
```

This gives 1.1E17, because the first .2 is entered as hwdouble type. Same happens when enter a floating point for a program.

So a question might be if input for programs can be always decimal, and casting to one  HwFP type happens during execution, if HwFP is set.

c3d, 2024-12-17:
This is intended behavior. Hardware floating-point uses a binary representation, which cannot convert exactly to decimal. Therefore, what you enter as decimal is converted to the closest binary floating point value. The documentation illustrates that with 0.2, which has no exact representation in binary because it's 1/5, and 5 is not a power of 2.

@skai111: The helpfile is already amended in the section called [Binary floating-point](https://github.com/c3d/db48x/blob/stable/help/db48x.md#binary-floating-point), which is part of the help about _Decimal numbers_, with an explanation that reads:

> The primary drawback of this representation is that it cannot represent some
> decimal values exactly, in the same way as 1/3 cannot be represented exactly
> in decimal. For example, 0.2 cannot be represented exactly using a binary
> floating-point representation.
>
> Using a binary format is not recommended if you need exact results on decimal
> values, for example adding monetary amounts. As an illustration, if you enable
> 16-digit hardware binary floating-point and add 0.20 and 0.45, the result is
> 0.65000 00000 00000 022. This is not a bug, but a limitation of the
> floating-point representation. The same computation with
> `SoftwareFloatingPoint` gives the exact expected result 0.65.

What could make sense, however, is to add a special prefix that makes it possible to force parsing in decimal or binary explicitly, in case you want to enter a program that contains both, as in the example given by @prismv.

So relabeling this as a feature and changing the title.

c3d, 2024-12-27:
After thinking a while about the various options, I think that the best solution is to visually differentiate the three numerical representations, by adding a `F` or `D` suffix for `float` and `double` floating-point values. This is inspired by the `f` suffix used in C for `float` values.

In other words:

* `1.2` always denotes a variable-precision decimal constants
* `1.2F` always denotes a 32-bit (`float`) IEEE754 value
* `1.2D` always denotes a 64-bit (`float`) IEEE754 value

These notations work both for input and output.

prismv, 2024-12-28:
Decimal dot is needed for input, e.g. `99.D`, which could be mentioned in docs.

Thank you for this fix, @c3d.

skai111, 2024-12-28:
That is  a very nice solution.
Thank you

commit d3ac61e8 (2024-12-27): hwfp: Add explicit `F` or `D` suffix

This commit introduces an explicit syntax with a `D` or `F` suffix for
binary IEEE754 floating-point values in 64-bit and 32-bit respectively.
In other words, `1.5` is always a (variable-precision) decimal value,
`1.5F` is always a 32-bit IEEE754 binary floating-point value, and
`1.5D` is always a 64-bit IEEE754 binary floating-point value. `F`
stands for `float` and `D` for `double`, the name of these types in
most modern C implementations.

The issue described in #1394 highlighted that the way a program was
parsed before depended on the `HardwareFloating [... cut]
commit 800bda19 (2024-12-29): hwfp: Add explicit `F` or `D` suffix

This commit introduces an explicit syntax with a `D` or `F` suffix for
binary IEEE754 floating-point values in 64-bit and 32-bit respectively.
In other words, `1.5` is always a (variable-precision) decimal value,
`1.5F` is always a 32-bit IEEE754 binary floating-point value, and
`1.5D` is always a 64-bit IEEE754 binary floating-point value. `F`
stands for `float` and `D` for `double`, the name of these types in
most modern C implementations.

The issue described in #1394 highlighted that the way a program was
parsed before depended on the `HardwareFloating [... cut]
closed 2024-12-30 by c3d