c3d/db48x #1398: corruption due to state loading of default Demo.48S file

prismv, 2024-12-20:
Simulator 0.8.8-18-9271 aka 0.8.9

Demo.48S from dev branch is used. This issue does not happen at start of simulator, when loading this state file, but only, if _original_ state file from _repository_ is loaded from Setup->State menu.

Thereafter input of `9_° SIN` gives e.g. -0.00295 69957 92 (results differ in fact after every state file load).

If some characters are removed from state file, this issue does not happen. This is one possibility:

``` diff
@@ -1,9 +1,7 @@
 Directory {
 
 @@ ===========================================================================
-@@
 @@     Physics
-@@
 @@ ===========================================================================
 
 Physics directory {
```

Tests were also done with config/db48x.48k set in keymap.cfg.

This runtime / memory / whatever corruption is very likely also the reason for reset of keymap settings due to actual version of Demo.48S as reported in #1387.

It seems that this issue takes also place, if 

``` C++
        for (unicode c = prog.get(); c; c = prog.get())
        {
            byte buffer[4];
            size_t count = utf8_encode(c, buffer);
            rt.insert(bytes, buffer, count);
            bytes += count;
        }
    }
```
in sysmenu.cc is still effective, but `object::result exec = cmds->run();` (and dependent parts) are commented out.

prismv, 2024-12-21:
I checked this on another PC that should have very similar setup (only related difference I am aware of is X11 vs Wayland usage). But there the reported issue was ___not___ _reproducible_.

Therefore I close this issue (until I find a reason with interest for others when/why/how this can happen)

closed 2024-12-21 by prismv

c3d, 2024-12-30:
I suspect that you ran into the bug fixed by this commit: https://github.com/c3d/db48x/commit/873c11bdd301311154a21f2cbd81689fbaef9b0f

If that is the case, what happened is that a garbage collection was triggered between the time we enter `handle_function` and the time `obj->evaluate()` on line 5603. In most cases, the garbage collection happens due to the Undo stack / LastArg saves around line 5593