c3d/db48x #1399: Equation Library: close analysis of bug c#1 in "Plate Capacitor"

Wiljea, 2024-12-28:
After a long computation (10 s) MSOLVER gives "No solution?" for:
``` C=25_μF  εr=2.26  A=1_cm^2  Q=75_μC
'ROOT(ⒺPlate Capacitor;[d;σ;Ein;ΔV];[1_cm;1_(μC/m^2);1_(N/C);1_V])'  ```

where it is the first of the 4 eqns which turns out to be the cause of the problem:
```    "  '(C_μF)=Ⓒε0*εr*(A_(cm^2))/(d_cm)' "'convert(Ⓒε0*εr*(A_(cm^2))/(C_μF);1_cm)'
    "  '(ΔV_V)=(Ein_(V/m))*(d_cm)' "
    "  '(Ein_(N/C))=(σ_(μC/cm^2))/(Ⓒε0*εr)' "
    "  '(σ_(μC/m^2))=(Q_μC)/(A_(cm^2))' "```

Indeed, if we isolate by hand and calculate ```(d_cm)``` using:
``` 'convert(Ⓒε0*εr*(A_(cm^2))/(C_μF);1_cm)' ``` => ``` d=8.00418 57882 3⁳⁻⁹ cm ``` 

and giving its value as input to search only for the 3 remaining unknowns, it works:
``` C=25_μF  εr=2.26  A=1_cm^2  Q=75_μC d=8.00418 57882 3⁳⁻⁹ cm
@ Expecting [ σ=750 000. μC/m↑2 Ein=3.74803 89378 4⁳¹⁰ N/C ΔV=3. V ]
'ROOT(ⒺPlate Capacitor;[σ;Ein;ΔV];[1_(μC/m^2);1_(N/C);1_V])' ```

Otherwise, solving the first equation alone for ```(d_cm)``` is even problematic because the numerical answer doesn't make sense:
``` C=25_μF  εr=2.26  A=1_cm^2
@ Expecting [ d=24 157 817. cm ]  
@ Failing [ d=8.00418 57882 3⁳⁻⁹ cm ] 
'ROOT([(C_μF)=Ⓒε0*εr*(A_(cm^2))/(d_cm)];[d];[1_cm])' ```

The same type of hallucination appears if we modify the equation by isolating a priori directly for the unknown ```(d_cm)``` :
``` C=25_μF  εr=2.26  A=1_cm^2
@ Expecting [ d=-2.⁳⁻²³ cm ]  
@ Failing [ d=8.00418 57882 3⁳⁻⁹ cm ] 
'ROOT([(d_cm)=Ⓒε0*εr*(A_(cm^2))/(C_μF)];[d];[1_cm])' ```

At this point I can't go any further and I don't see any other clues that point to the root cause of the problem.

c3d, 2025-01-17:
Reduced test case:

```
C=25_μF  εr=2.26  A=1_cm^2  Q=75_μC
'ROOT((C_μF)=Ⓒε0*εr*(A_(cm^2))/(d_cm);d;1_cm)'
```

c3d, 2025-01-17:
Recommendation: use a more credible initial value for `d`, like `1_pm`.

commit 352cfe5f (2025-01-17): equations: Change initial guess for Plate Capacitor

The initial guess for `d` was so large that the algorithm fails to
converge with the default number of iterations allowed.

The situation is similar to solving `1-1/x` with an initial value of
`1E9`: the area of the function is so flat that far out that following
the slope leads you to a large negative value, making things worse. It
then takes a while to reconverge. Ultimately, we evaluate the function
more than `SolverIterations` times, so the solver emits `No solution`.

The HP50G and HP48SX behave even worse with similar input: they just
 [... cut]
closed 2025-02-03 by c3d