c3d/db48x #1400: Equation Library: close analysis of bug c#2 in "Drift Speed & Current Density"

Wiljea, 2024-12-28:
Here the problem arise (trial #1) in the preliminary algebraic computations of input values before the call of ROOT, which otherwise works correctly if the right input values are pre-calculated (trial #2):
```rpl
@ C#2 First trial #1 where the preliminary calculations of input variables A, ρ & σ doesn't work "Invalid algebraic" BUT algebraics: OK
I=1_A  n=8.5e28_(m^-3)  r=0.1_cm  A='Ⓒπ*(r_cm)^2'  ρ='Ⓒqe*n' σ='CONVERT(→NUM(Ⓒqe*(n_(m^-3))*40_(cm^2/(V*s)));1_S/m)'
@ Failing [ vd=2.33733 41683 6⁳⁻⁵ m/s J=31.83098 86184 A/cm↑2 E=5.84333 54209⁳⁻³ V/m ]
'ROOT(ⒺDrift Speed & Current Density;[vd;J;E];[1_m/s;1_(A/cm^2);1_(V/m)])'
```
```rpl
@ Second working trial #2 where the input variables A, ρ & σ are a priori calculated with
@ A='Ⓒπ*(0.1_cm)^2'  ρ='Ⓒqe*n'  σ='CONVERT(Ⓒqe*n*40_(cm^2/(V*s));1_S/m)' 
I=1_A  n=8.5e28_(m^-3) A=3.14159 26535 9⁳⁻² cm↑2 ρ=1.36185 01389E10_C/m↑3 σ=54 474 005.556_S/m
@ Expecting [ vd=2.33733 41683 6⁳⁻⁵ m/s J=31.83098 86184 A/cm↑2 E=5.84333 54209⁳⁻³ V/m ]
'ROOT(ⒺDrift Speed & Current Density;[vd;J;E];[1_m/s;1_(A/cm^2);1_(V/m)])'
```
It is worth mentioning that the algebraic calculation of σ only works with
```rpl
'CONVERT(→NUM(Ⓒqe*(n_(m^-3))*40_(cm^2/(V*s)));1_S/m)'
```
=> σ=54 474 005.556_S/m
whereas (without "→NUM") the following fails with "Bad argument type"
```rpl
'CONVERT(Ⓒqe*n*40_(cm^2/(V*s));1_S/m)'
```

Therefore, it seems that it is the algebraic calculation of  (```σ=54 474 005.556_S/m```) which is problematic.

c3d, 2025-01-19:
There were several combined issues that are now all fixed:

1. When doing a subtraction, the "subtraction of units" path would only be taken
   if both sides were unit objects. This failed if one side was an
   expression, resulting in `Inconsistent units` message. This is similar to the
   recently fixed problem with `Convert`, but for `+` and `-`. This applied for
   the subtraction between left and right terms in the first equation selected.

2. When one of the variables contained an expression, for example
   `A='Ⓒπ*(0.1_cm)^2'` which assigned an expresssion to `A`, the evaluation of
   that expression was deferred until after the solver needed to use it, because
   of the run loop deferral optimization. The solution is to force the run loop
   to complete for that call stack depth when the `sync` flag is set. This
   caused the `Invalid algebraic` error.

3. A second, different `Inconsistent units` resulted from the initial value for
   `A` being incorrectly parsed due to a test harness problem. Where the
   initial value was set as `A='Ⓒπ*(0.1_cm)^2'`, the RPL engine ended up
   receiving `A='π*(0.1_cm)^2'`, i.e. symbol instead of constant. This was due
   to a bug in the assumptions for the code transmitting a UTF8 sequence, which
   was intended to stop at the end of a UTF8 glyph (the buffer being sized
   accordingly) but would instead accumulate multiple UTF-8 sequences until the
   buffer size was reached. This led to incorrect byte sequences being sent.

Wiljea, 2025-02-06:
Done.

closed 2025-02-06 by Wiljea